mperrin/webbpsf #34: New SIAFs for webbpsf-data package

josePhoenix, 2015-01-15:
Upon investigating the current public SIAFs, it appears the SIAF parsing code in `jwxml.py` now expects a slightly different format. As far as I know, WebbPSF doesn't use SIAFs in its calculations (yet!) but we should update them anyway. (That way when field dependence support lands, it'll require a code update but not [necessarily] a data update.)

mperrin, 2015-01-15:
We should ping Colin Cox on this. I believe he has already made the relevant updates in his version of the SIAF reading code.

 - Marshall

josePhoenix, 2015-01-15:
I've got the new XML files from Colin; however, he warned me that he would be updating several of them in the near future. `jwxml` seems to be able to read them (but not the ones currently in webbpsf-data).

mperrin, 2015-01-15:
Ah OK I misread the issue. I thought you were saying the code needed to be updated but you mean the data files. By all means yes let's go ahead and add the current ones now, and we can update again before the release if Colin makes another updated set.

josePhoenix, 2015-01-16:
All right, I think that's up to you since I can't write to the folder. Per Colin, the updates files are in `/itar/jwst/tel/share/SIAF_WG/Instruments/SIAFXML/XML/`.

mperrin, 2015-01-21:
I have updated the files in `/grp/jwst/ote/webbpsf-data` with Colin's latest SIAF XML files. (and my master source directory as well, so whenever I next generate a new zip file the updated XML will be included in that)

Reminder to myself:
 - [ ]  Relocate master copy of webbpsf data to `/itar/jwst/tel` and set up shared permissions
 - [ ]  Document build scripts for making the webbpsf data zip files
 - [x] Ping Dean about the MIRI filter files.

mperrin, 2015-02-13:
Master files relocated. (Again there is not actually any ITAR content in them, this is just the path to use given the existing shared files area on that disk)

Zip file building scripts updated; new zip files made for this release. I had to bump the version number a bit since the filter files changed, so it's now 0.3.1. I am slightly dismayed to have different version number on the data file than on the release itself but I don't think it can be helped right now.

closed 2015-02-13 by mperrin