mperrin/webbpsf #35: potential issue with parity of forward & inverse MFT transforms

mperrin, 2015-01-15:
Begin forwarded message:

> From: Anand Sivaramakrishnan <anand@stsci.edu>
> Subject: troubles with parity of forward & inverse mDFT transforms...
> Date: January 15, 2015 at 2:42:48 PM EST
>
> Hi Joseph & Marshall
>
>I use matrixDFT from WebbPSF as a standalone (to try to unify my code with WebbPSF utilities), and I'm having some trouble with coordinate flips when transforming forwards, then back.  
>
>I start with a psf in this example, transform it forwards to visibilities, then I try to transform it backwards as well as forwards - but get a flipped PSF both times.  Running "matrixDFT_test.py" in the attached tar file will do these things, and give you two convenient ds9 commands to cut'n'paste to see the results.
>
> This is a bit worrying, as matrixDFT is at the core of many things I do.
>
> I also include matrixDFT.py copied from WebbPSF, with a print statement to show that I'm actually using this file from the *_test.py  driver.
> 
> Any comments or suggestions would be great!  If I should be using a different 'entry point' into the dft (such as SFT) please do let me know.

(I don't see any straightforward way to attach the file attachment to a github issue, but we have it in our inboxes)

mperrin, 2015-01-21:
Reproduced by @josePhoenix from Anand's example code. Definitely looks like I missed this one somehow before.

josePhoenix, 2015-01-23:
Really a POPPY issue, see https://github.com/mperrin/poppy/issues/38 for remaining subtasks (basically writing test cases).

closed 2015-01-23 by josePhoenix