mperrin/webbpsf #36: webbpsf.JWInstrument.calcPSF duplicates code from poppy.Instrument.calcPSF

josePhoenix, 2015-01-23:
We should factor out the JWST-specific stuff where possible and call the superclass' calcPSF.

josePhoenix, 2016-04-13:
@mperrin It looks like the only things that will change if I remove `SpaceTelescopeInstrument.calcPSF` entirely are:

  * no more `calc_oversample=` argument to raise a deprecation warning (we said we would remove this in 0.4)
  * no more `filter=` argument to override `Instrument.filter` -- I think this is a good change for consistency with `poppy.Instrument`, and I wasn't aware it was there

mperrin, 2016-04-14:
I agree those are fine to remove. The other small detail (also OK to remove) is the code I added to set the default source spectrum to a 5700 K blackbody, to silence the warning or error that ``poppy.Instrument._getWeights`` would otherwise produce. The more I think about it, the less I think that should be a warning message, since it's such a common usage case for users to ask for a PSF without giving an explicit source spectrum.   

So I would propose in ``poppy`` we convert that to a benign info message instead of a warning, and then there's no need for any code in ``webbpsf.Instrument`` to avoid triggering it.  I think I may just go ahead and do that right now.

mperrin, 2016-12-05:
This was fixed in #132. There is no longer a separate `JWInstrument.calcPSF` (nor `calc_psf`); it just uses the one in `poppy`.

closed 2016-12-05 by mperrin