mperrin/webbpsf #40: Jitter functionality is incomplete

josePhoenix, 2015-01-27:
In the documentation for WebbPSF.JWInstrument, we have

```
    jitter : string "gaussian" or None
        Type of jitter model to apply. Currently only convolution with a Gaussian kernel of specified
        width `jitter_sigma` is implemented. (default: None)
    jitter_sigma : float
        Width of the jitter kernel in arcseconds (default: 0.007 arcsec)
```

However, this test code shows no difference between a PSF with jitter enabled and a PSF with no jitter:
```python
import webbpsf
nc = webbpsf.NIRCam()
nc.options['jitter'] = None
psf = nc.calcPSF(nlambda=3)

nc.options['jitter'] = 'gaussian'
psf_jitter = nc.calcPSF(nlambda=3)
webbpsf.display_PSF_difference(psf, psf_jitter)
```

I think this is because calcPSF is completely rewritten for `JWInstrument` and doesn't automatically get updates we make to `poppy.Instrument`. Refactoring that is something to tackle for v0.4.

mperrin, 2015-02-16:
Does #47 fix this?

josePhoenix, 2015-02-19:
Yes!

*Achievement unlocked: we closed all the issues on a milestone.*

closed 2015-02-19 by josePhoenix