mperrin/webbpsf #46: WFIRST test case fails if no pysynphot is present

mperrin, 2015-02-13:
The WFIRST imager class fails for the fallback code for the no-pysynphot case.  Looks like something minor in terms of the filter file setup or header, hopefully easy to fix. 

I'm going to make it xfail for now, in the hopes of almost having the tests actually pass on Travis.

```
webbpsf/tests/test_wfirst.py:3: test_wfirstimager_psf FAILED

====================================================================================== FAILURES =======================================================================================
________________________________________________________________________________ test_wfirstimager_psf ________________________________________________________________________________

    def test_wfirstimager_psf():
        """Just test that instantiating WFIRSTImager works and can compute a PSF"""
        wi = wfirst.WFIRSTImager()
>       wi.calcPSF()

webbpsf/tests/test_wfirst.py:6:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <webbpsf.wfirst.WFIRSTImager object at 0x1101dd610>, outfile = None, source = None, filter = None, nlambda = 10, monochromatic = None, fov_arcsec = 5.0, fov_pixels = None
oversample = 4, detector_oversample = 4, fft_oversample = 4, calc_oversample = None, rebin = True, clobber = True, display = False, return_intermediates = False, kwargs = {}
local_options = {'detector_oversample': 4, 'fft_oversample': 4, 'monochromatic': None, 'nlambda': 10}, fov_spec = 'arcsec = 5.000000'

    def calcPSF(self, outfile=None, source=None, filter=None, nlambda=None, monochromatic=None,
                fov_arcsec=None, fov_pixels=None,  oversample=None, detector_oversample=None,
                fft_oversample=None, calc_oversample=None, rebin=True, clobber=True, display=False,
                return_intermediates=False, **kwargs):
        """ Compute a PSF.

            The result can either be written to disk (set outfile="filename") or else will be returned as
            an astropy.io.fits HDUList object.

            Output sampling may be specified in one of two ways:

            1) Set `oversample=<number>`. This will use that oversampling factor beyond detector pixels
               for output images, and beyond Nyquist sampling for any FFTs to prior optical planes.
            2) set `detector_oversample=<number>` and `fft_oversample=<other_number>`. This syntax lets
               you specify distinct oversampling factors for intermediate and final planes. This is generally
               only relevant in the case of coronagraphic calculations.

            By default, both oversampling factors are set equal to 4. This default can be changed in your
            webbpsf configuration file.

            Notes
            -----
            More advanced PSF computation options (pupil shifts, source positions, jitter, ...)
            may be set by configuring the `.options` dictionary attribute of this class.

            Parameters
            ----------
            filter : string, optional
                Filter name. Setting this is just a shortcut for setting the object's filter first, then
                calling calcPSF afterwards.
            source : pysynphot.SourceSpectrum or dict or tuple
                specification of source input spectrum. Default is a 5700 K sunlike star.
            nlambda : int
                How many wavelengths to model for broadband?
                The default depends on how wide the filter is, as set by a lookup table in the webbpsf data distribution.
            monochromatic : float, optional
                Setting this to a wavelength value (in meters) will compute a monochromatic PSF at that
                wavelength, overriding filter and nlambda parameters.
            fov_arcsec : float
                field of view in arcsec. Default=5
            fov_pixels : int
                field of view in pixels. This is an alternative to fov_arcsec.
            outfile : string
                Filename to write. If None, then result is returned as an HDUList
            oversample, detector_oversample, fft_oversample : int
                How much to oversample. Default=4. By default the same factor is used for final output
                pixels and intermediate optical planes, but you may optionally use different factors
                if so desired.
[... 173 lines cut ...]
            # compute a source spectrum weighted by the desired filter curves.
            # The existing FITS files all have wavelength in ANGSTROMS since that is the pysynphot convention...
            filterfits = fits.open(self._filters[self.filter].filename)
            filterdata = filterfits[1].data
            try:
                f1 = filterdata.WAVELENGTH
                d2 = filterdata.THROUGHPUT
            except:
                raise ValueError("The supplied file, %s, does not appear to be a FITS table with WAVELENGTH and THROUGHPUT columns." % self._filters[self.filter].filename )
            if 'WAVEUNIT' in  filterfits[1].header.keys():
                waveunit  = filterfits[1].header['WAVEUNIT']
            else:
                poppy_core._log.warn("CAUTION: no WAVEUNIT keyword found in filter file {0}. Assuming = Angstroms by default".format(filterfits.filename()))
                waveunit = 'Angstrom'
>           if waveunit != 'Angstrom': raise ValueError("The supplied file, %s, does not have WAVEUNIT = Angstrom as expected." % self._filters[self.filter].filename )
E           ValueError: The supplied file, /Users/mperrin/software/webbpsf-data/WFIRSTImager/filters/Z087_throughput.fits, does not have WAVEUNIT = Angstrom as expected.

/Users/mperrin/Dropbox (Personal)/Documents/software/poppy/poppy/instrument.py:722: ValueError
----------------------------------------------------------------------------------- Captured stderr -----------------------------------------------------------------------------------
poppy     : INFO     Optic from /Users/mperrin/software/webbpsf-data/AFTA_symmetrical.fits: Loaded amplitude transmission from /Users/mperrin/software/webbpsf-data/AFTA_symmetrical.fits
webbpsf   : INFO     Setting up PSF calculation for WFIRSTImager
webbpsf   : WARNING  Filter Z087 not found in lookup table for default number of wavelengths to use.. setting default nlambda=10
poppy     : WARNING  Pysynphot unavailable (or invalid source supplied)!   Assuming flat # of counts versus wavelength.
================================================================== 1 failed, 36 passed, 1 xfailed in 361.24 seconds ==================================================================
```

josePhoenix, 2015-02-13:
Ah, this is just a case of making the comparison case insensitive. I'll open a pull request to POPPY.

josePhoenix, 2015-02-13:
Fixed through changes in POPPY.

closed 2015-02-13 by josePhoenix