mperrin/webbpsf #51: Set up Travis test with pysynphot and some minimal data file set

mperrin, 2015-02-19:
I was reminded by #50 about an earlier suggestion. We should set up a Travis test instance that installs Pysynphot plus some bare-minimum set of FITS files (which I think could potentially be just some single stellar atmosphere model instance perhaps?). This would ensure test coverage of everything both with and without pysynphot.

josePhoenix, 2016-02-04:
Putting this where it makes more sense:

A stupid hack to find all the files pysynphot touches. As part of the test case for #98, I thought I could maybe monkeypatch the `rootdir` module-level attribute in `pysynphot.locations`, but it gets used to define some other variables in that module at import time. My next plan was re-importing pysynphot, but it wasn't picking up the modified `os.environ` variables. Current plan is to identify a minimal set of files to run the test suite using aforementioned stupid hack.

```
PYSYN_CDBS=/grp/hst/cdbs/ strace python -c "import pysynphot; pysynphot.ObsBandpass('wfc3,im,f120w')" 2>&1 | grep "/grp/hst/cdbs/" | grep "stat("
stat("/grp/hst/cdbs/extinction", {st_mode=S_IFDIR|S_ISGID|0755, st_size=1024, ...}) = 0
stat("/grp/hst/cdbs/extinction", {st_mode=S_IFDIR|S_ISGID|0755, st_size=1024, ...}) = 0
stat("/grp/hst/cdbs/extinction/milkyway_rv4_001.fits", {st_mode=S_IFREG|0644, st_size=8640, ...}) = 0
stat("/grp/hst/cdbs/extinction/xgal_starburst_001.fits", {st_mode=S_IFREG|0654, st_size=40320, ...}) = 0
stat("/grp/hst/cdbs/extinction/milkyway_dense_001.fits", {st_mode=S_IFREG|0654, st_size=8640, ...}) = 0
stat("/grp/hst/cdbs/extinction/lmc_diffuse_001.fits", {st_mode=S_IFREG|0654, st_size=8640, ...}) = 0
stat("/grp/hst/cdbs/extinction/lmc_30dorshell_001.fits", {st_mode=S_IFREG|0654, st_size=8640, ...}) = 0
stat("/grp/hst/cdbs/extinction/milkyway_diffuse_001.fits", {st_mode=S_IFREG|0654, st_size=8640, ...}) = 0
stat("/grp/hst/cdbs/extinction/milkyway_rv21_001.fits", {st_mode=S_IFREG|0644, st_size=8640, ...}) = 0
stat("/grp/hst/cdbs/extinction/smc_bar_001.fits", {st_mode=S_IFREG|0654, st_size=8640, ...}) = 0
stat("/grp/hst/cdbs/mtab/yah1742qm_tmg.fits", {st_mode=S_IFREG|0664, st_size=492480, ...}) = 0
```

mperrin, 2016-02-04:
also moving this comment from #98:

Well, since webbpsf contains its own reference tables for the normalized JWST instrument throughputs that it turns into ObsBandpasses, we don't actually need any instrument graph tables. (At one point years ago draft versions of these existed for the JWST instruments, but I don't know if Pandeia even uses those internally any more.)

It would be convenient to grab one or two stellar atmosphere model files (e.g. a file or two from the phoenix model grid) so we can make a test case using those. Though we could also cut that out and make test cases using black body spectra.

The real question is, what minimalist file set is needed so that pysynphot doesn't just immediately give up on import, a la TypeError: initializing GraphTable with GFile=None; possible bad/missing CDBS.

mperrin, 2016-02-04:
And I realize now, maybe we can even work around that. Looking at the code I'm reminded that `webbpsf` first attempts to request an `ObsBandpass`  via  e.g. `band=ObsBandpass('nircam,im,f200w')`, then it falls back to using its own reference files via `band=ArraySpectralElement(arrays_from_ref_files_in_WEBBPSF_DATA)`. 

Thing is, that `ObsBandpass` call is going to work for precisely _no one_.  Not even me any more since I don't have the old dev JWST-inclusive CDBS as my active `$PYSYN_CDBS` any more.

Suggestion: We remove that `ObsBandpass` call and then `webbpsf` will *always* just use its own internally provided reference files. No need for any graph table at all. 

And then (separately) we actually finish implementing the build script to extract normalized JWST SI throughput tables via calls to Pandeia's `instrument.get_total_eff()` function, and we use that whenever we make a new set of webbpsf reference files.

josePhoenix, 2016-02-08:
Okay, that sounds reasonable. We should have some code that exercises pysynphot-based source SEDs, though.

mperrin, 2016-02-08:
Yes - but we could just use Planck function or power law SEDs in pysynphot, which shouldn't require any input data files.

mperrin, 2018-08-28:
Issues migrated to https://github.com/spacetelescope/webbpsf/issues. 
Please see the copies in that repo from now on.

closed 2018-08-28 by mperrin