mperrin/webbpsf #56: Verify Pupil Shift GUI control does what we want it to

josePhoenix, 2015-02-23:
I'm not sure what the expected behavior is, but I tried a variety of different values for pupil shift in the NIRCam tab and didn't observe any change in the PSF. Only when using semi-analytic coronagraph propagation did I see evidence of the values I entered in the terminal output:

```
webbpsf   : INFO     Creating optical system model:
poppy     : INFO     Initialized OpticalSystem: JWST+NIRCam
poppy     : INFO     JWST Pupil: Loaded amplitude transmission from /Users/jdl/dev/webbpsf-data/pupil_RevV.fits
poppy     : INFO     JWST Pupil: Loaded OPD from  /Users/jdl/dev/webbpsf-data/NIRCam//OPD/OPD_RevV_nircam_132.fits, plane 0
poppy     : INFO     Added pupil plane: JWST Pupil
poppy     : INFO     Added image plane: MASK210R
poppy     : INFO     CIRCLYOT: Loaded amplitude transmission from /Users/jdl/dev/webbpsf-data/NIRCam//optics/NIRCam_Lyot_Somb.fits
poppy     : INFO     Requested optic shift of ( 0.100,  0.000) %
poppy     : INFO     Actual shift applied   = ( 0.100,  0.000) %
poppy     : INFO     Added pupil plane: CIRCLYOT
poppy     : INFO     Added detector: NIRCam detector, with pixelscale=0.031700 arcsec/pixel and oversampling=2
webbpsf   : INFO     Trying to invoke switch to Semi-Analytic Coronagraphy algorithm
webbpsf   : INFO     SAC OK
```

That was with 10% entered in the "X" box and 0 in the "Y". It's not clear whether it's being divided by 100 twice from the "( 0.100,  0.000) %" line.

mperrin, 2015-02-23:
You can most easily see the expected behavior if you display the optical system of the instrument. You should see the intermediate pupil plane shift with respect to the telescope primary. 

The "( 0.100, 0.000) %" line I suspect is just an incorrect output string which doesn't need the percent sign. This functionality definitely used to work properly for shifts of a few percent so I suspect it's just a poorly written string rather than a repeated division.

mperrin, 2018-08-28:
Issues migrated to https://github.com/spacetelescope/webbpsf/issues. 
Please see the copies in that repo from now on.

closed 2018-08-28 by mperrin