mperrin/webbpsf #58: MIRI cannot switch to SemiAnalyticCoronagraph when using Lyot masks

josePhoenix, 2015-02-26:
Problem identified by @kvangorkom. Example + output in https://gist.github.com/kvangorkom/808d17521ed9f99076e5

```
webbpsf : INFO Trying to invoke switch to Semi-Analytic Coronagraphy algorithm
webbpsf : WARNING Could not switch to Semi-Analytic Coronagraphy mode; invalid set of optical planes? Using default propagation instead.
webbpsf : WARNING Input optical system must have exactly 4 planes to be convertible into a SemiAnalyticCoronagraph
```

mperrin, 2015-02-26:
There's two different things going on here - first, the SAC class doesn't know what to do with the rotation plane in the optical system.  We could probably work around that if desired by extending the code: It should just do the SAC calculation for the first 3 optical planes, then rotate the PSF by the necessary 4.5 degrees after that. 

But more fundamentally, the SAC approach isn't all that well suited to the MIRI Lyot because the Lyot occulter is held up on a narrow wire extended across the coronagraph field, and there's an outer field stop as well. So the fundamental assumption of the SAC method doesn't hold true that we only have a small central occulter and therefore can only calculate the PSF in the intermediate plane over that field.   In other words, we'd need to calculate the intermediate PSF over a large enough FOV that there's probably no speed advantage to the semi analytic approach.  

At some point I must have thought it made sense to try the SAC mode for MIRI, but I'm not entirely convinced any more that's actually the best approach.  Certainly if you care about the outer edges of the FOV set by the field stop in MIRI then the SAC approach doesn't work at all. 

Simple experiment: comment out the 4.5 deg rotation, and see what the relative speeds are for MIRI Lyot calculated both ways.

mperrin, 2018-08-28:
Issues migrated to https://github.com/spacetelescope/webbpsf/issues. 
Please see the copies in that repo from now on.

closed 2018-08-28 by mperrin