mperrin/webbpsf #63: Optimal offsets along NIRCam bar occulters

lajoiecp, 2015-04-14:
I'd like to suggest the implementation of a method to readily offset the source along the bar occulters based on the filter wavelength.  I have created my own (based on internal documentation I could get a hold of) and would gladly provide it for a start.

mperrin, 2015-04-14:
Yes, agreed. 

Do we know how exactly this is going to be supported in APT yet? Are there going to be specific default positions for each filter specified in the SIAF, is it all going to be done with POS TARG offsets, or something else? We should definitely implement something along the lines you describe, and hopefully it can be reasonably consistent with the eventual APT interface as well.

lajoiecp, 2015-04-15:
John S. provided me with one of his TR (to be released) describing the width of the bars as a function of displacement along the bars, as well as the optimal position (i.e. where the width is 4 lambda/D) for a given wavelength. 

The bars are 20" wide with flat edges.  His equations for the middle 15" are linear, going from 2 lambda/D at the narrow end of the bar to 6 lambda/D at the wide end. The centers of the bars are 4 lambda/D at 2.10 and 4.6 um, which seems consistent with the WebbPSF setup.

Not exactly sure how this is going to be setup in APT, but from what I understand from talking to John, it will (or should) be offered to APT users.

mperrin, 2016-04-27:
I believe there are now updated locations in the SIAF for each of the filter-dependent bar occulter positions?  we should put in something consistent with those. In any case @Skyhawk172 if you can give me a copy of the code you mentioned a year ago here (or anything more recent?) then we can fold it into the next release of webbpsf. thanks!

mperrin, 2016-04-28:
Sent to me by email, copied here for reference: 

```
def get_source_offset(lambda0, filt, mask_coron):
    ########################################################
    # Find optimal position for source behind bar occulters.
    # See document from John Stansberry (email 04/08/15) 
    # for equations below; modified for orientation of bars.
    #
    # Position (x,y)=(0,0) for both SWB and LWB are 4 lambda/D 
    # wide for 2.10 and 4.60 um respectively. Use scaling below  
    # to offset for other wavelengths. Results are arcseconds.
    #
    # Bars are 20" wide with flat edges. Linear part is 15" wide.
    # 
    # Width of bar occulters as a function of x, where x=0
    # is the center of the bars:
    # WSWB(x) (") = 0.2666" + 0.01777 * x
    # WLWB(x) (") = 0.5839" + 0.03893 * x
    ########################################################

    y0 = 0
    if mask_coron == 'MASKSWB':
        if filt[-1]=="W": x0 =-6.83*( np.sqrt(lambda0) - 2.196 )
        else:             x0 =-7.14*( lambda0 - 2.10 )
    elif mask_coron == 'MASKLWB':
        if filt[-1]=="W": x0 =-3.16*( np.sqrt(lambda0) - 4.747 )
        else:             x0 =-3.26*( lambda0 - 4.60 )
    else: x0,y0 = 0.,0.

    return x0,y0

```

JarronL, 2017-03-02:
I believe the equation for `get_source_offset()` is incorrect for the wide-band filters. The wavelength value is not meant to be inside a square root. For the wide filters, lambda0 is the "half-transmission point, lambda^(1/2)" according to NIRCam_CoronagraphOps_V4.pdf. The variable lambda^(1/2) is misleading in the sense that the "^(1/2)" is not to the 0.5 power, but a confusingly placed superscript.

mperrin, 2017-03-13:
Thanks for pointing out that error, Jarron. I've just asked John Stansberry to send me the specific table of offsets along the bar occulter as implemented in the SIAF tables. I'll just use that lookup table rather than the equation and that should ensure strict consistency in any case. 

This raises the question of what the API should look like for setting offsets along the bar. Should it be the case that WebbPSF automagically applies offsets along the bar occulters based on filter name? I.e. if you do
```
nrc = webbpsf.NIRCam()
nrc.image_mask = 'MASKLWB'
nrc.pupil_mask= 'WEDGELYOT'
nrc.filter = 'F460M'
```
then behind the scenes it applies the appropriate offset? Or would it be better to make that an explicit function call:

```
nrc.apply_bar_offset('MASKLWB', 'F460M')
```

?

@kvangorkom do  you have any thoughts on what you'd like to see in an interface for this?

mperrin, 2017-03-13:
From Stansberry: 
```
NIRCam **Module-A** Bar-occulter offset vs. filter

output from filt_baroffset.pro at Mon Mar 13 15:34:54 2017

Offset is positive in the direction of increasing 
X_raw pixel position

Offsets are approximated here as being only in the
X pixel coordinate (raw column #)
The current center position for the two BAR masks in
raw pixel coordinates are:
MASKSWBar  (SCA A4): [490.5, 542.5]
MASKLWBar  (SCA A5): [434.5, 1667.0]

                          Offset         Raw Pixel Position
                     from Bar Center          on SCA
Filter     Wavel    arcsec   dXpix_raw    Xraw      Yraw
-------------------------------------------------------------
Short Wavelength Bar (SCA A4)
 F182M      1.84    -1.856     -58.0     432.5     542.5
 F187N      1.88    -1.571     -49.1     441.4     542.5
 F210M      2.09    -0.071      -2.2     488.3     542.5
 F212N      2.12     0.143       4.5     495.0     542.5
 F200W      2.23     0.232       7.3     497.8     542.5

Long Wavelength Bar (SCA A5)
 F250M      2.50    -6.846    -105.3     329.2    1667.0
 F300M      2.99    -5.249     -80.7     353.8    1667.0
 F277W      3.14    -5.078     -78.1     356.4    1667.0
 F335M      3.35    -4.075     -62.7     371.8    1667.0
 F360M      3.62    -3.195     -49.2     385.3    1667.0
 F356W      3.97    -2.455     -37.8     396.7    1667.0
 F410M      4.09    -1.663     -25.6     408.9    1667.0
 F430M      4.28    -1.043     -16.0     418.5    1667.0
 F460M      4.63     0.098       1.5     436.0    1667.0
 F480M      4.79     0.619       9.5     444.0    1667.0
 F444W      4.99     0.768      11.8     446.3    1667.0
```

kvangorkom, 2017-03-14:
> @kvangorkom do you have any thoughts on what you'd like to see in an interface for this?

Hmm. The default behavior in APT/future-real-observations is to consider a star with no requested offset to be at this optimal position behind the bar, correct? If so, my intuition is that defaults that are consistent with these behaviors would be preferable. (A new user to WebbPSF would expect [0,0] in WebbPSF to match [0,0] in APT to match [0,0] in their observations.) So that's a vote for the automagic approach.

That said, it might throw off current users to discover the bars have suddenly have shifted under them. I don't know what the ideal trade-off is, but I'm inclined to favor an approach that doesn't require future users to know they're required to take an additional step to get WebbPSF to replicate the behaviors in these others tools and the observatory itself.

mperrin, 2017-03-14:
Yeah, I think I'm convinced that the default behavior should be as you describe. That changes things for current users, but for the better in the long run. We don't have a big enough user base that I worry so much about breaking back compatibility in something as specific as NIRCam bar coronagraphy. We can ask our expert coronagraphers to update their codes. :-) And we can make sure to signpost this clearly in the next set of release notes. 

Plots of offsets along the bars, based on the numbers above. I'm putting these here now so I remember to add them to some documentation page later.

[image]

[image]

mperrin, 2017-03-14:
Implemented. If you set up a coronagraphic observation using either bar, it will automatically offset along the occulter to the specified position for the currently selected filter. Seems to be working here, and the unit tests pass so I don't seem to have broken anything elsewhere in the code by adding this. However if anyone feels like kicking the tires on this any, I'd appreciate any test reports. 

This new behavior can be overridden by setting the `bar_offset` option manually. For instance
```
nrc=webbpsf.NIRCam()
nrc.options['bar_offset'] = 0
```
will make it revert to the prior behavior where all calculations are centered on the middle of the occulter. 

Note, this offset is _not_ implemented by tilting the star, rather it's done by offsetting the image mask transmission pattern itself. Thus the convention is retained that the on-axis star is at zero tip and tilt. This makes things simpler for integration with other tools such as Pandeia and @kvangorkom's pandeia-coronagraphy package.

mperrin, 2017-08-11:
Stale issue. Implemented back in the spring; no further action needed now.

closed 2017-08-11 by mperrin