mperrin/webbpsf #65: Incorporate wavelength limits when GR700XD is used with 'CLEAR' filter on NIRISS

josePhoenix, 2015-04-20:
Based on correspondence with André Martel, it sounds like configuring the filter wheel as `'CLEAR'` and the pupil wheel as `'GR700XD`' is a common use case. In this case, we should limit the calculation to wavelengths that are passed in that mode: namely, from `NIRISS.SHORT_WAVELENGTH_MIN` to `NIRISS.SHORT_WAVELENGTH_MAX` (0.5 - 2.35 um).

mperrin, 2015-04-20:
Relatedly we would want to have someone on the NIRISS team provide us with lookup tables for the GR700XD displacement vs wavelength, or else all those wavelengths would end up on top of one another in a non-useful blob. 

Thus far we've ignored that since the Pandeia team is putting together the monochromatic PSFs themselves, but it would be needed for any attempts at broadband simulations in dispersed modes. 

The other complication which will definitely need to be special-cased for GR700XD is that there are multiple orders transmitted through the grism, so there's no longer a one-to-one mapping to the output detector. We could implement a special instrument option to pick order 1 to 3 when the GR700XD is selected.

mperrin, 2015-04-20:
As discussed in person but writing this down for the record. The grism wavelength cutoff is imposed not by any optical transmission limit but rather the edge of the detector.  See [this TIPS/JIM talk on Confluence](https://confluence.stsci.edu/download/attachments/48727484/TIPS_2014_04_17_C.pdf?version=1&modificationDate=1413498576000&api=v2) in particular slides 20-21. 

Note incidentally the `SHORT_WAVELENGTH_MAX`/`LONG_WAVELENGTH_MIN` cutoff for NIRISS actually translates to `STUFF_WE_STUCK_IN_THE_PUPIL_WHEEL_MAX` / `STUFF_WE_STUCK_IN_THE_FILTER_WHEEL_MIN`.  It's all on the same detector, very different from the case of NIRCam.

mperrin, 2018-08-28:
Issues migrated to https://github.com/spacetelescope/webbpsf/issues. 
Please see the copies in that repo from now on.

closed 2018-08-28 by mperrin