mperrin/webbpsf #67: Remove non-specific `except` blocks

josePhoenix, 2015-04-24:
There are places in the WebbPSF and POPPY code where the

```
try:
   # do things
except:
   # fallback
```
anti-pattern occurs. All `except` blocks should say what kind of errors they expect. Otherwise, they can mask other errors. For example,

```python
        if nlambda is None or nlambda==0:
            # Automatically determine number of appropriate wavelengths.
            # Make selection based on filter configuration file
            try:
                nlambda = self._filter_nlambda_default[self.filter]
                _log.debug("Automatically selecting # of wavelengths: %d" % nlambda)
            except:
                nlambda=10
                _log.warn("Filter %s not found in lookup table for default number of wavelengths to use.. setting default nlambda=%d" % (self.filter, nlambda))
```
was hiding the fact that `self._filter_nlambda_default` didn't actually exist. That should be `except KeyError` to catch the case where the filter isn't in the lookup dict. (Or even, alternatively, `dict.get(<key>, <default>)`.)

mperrin, 2016-01-21:
At this point, this is mostly done except in GUI code (tkgui.py/wxgui.py). There was one non-specific except in jwxml that I just fixed. And there is one in wfirst.py. @josePhoenix could you take care of that one please? 

I am inclined to not worry much about fixing this in the GUI code, since we are gradually deprecating that in favor of notebooks.

mperrin, 2016-02-08:
Every non-specific except is fixed, except for a small handful in the `wxgui` code that we're in the process of slowly deprecating. That's good enough for me.

closed 2016-02-08 by mperrin