mperrin/webbpsf #80: Release checklist for 0.4

mperrin, 2015-11-19:
- [ ] Add WFIRST page to Sphinx docs @josePhoenix 
- [ ] Review WFIRST notebook one last time @mperrin
- [ ] Update (some) figures in Sphinx docs for better color tables consistency etc. 
- [ ] Regenerate updated filter tables for NIRISS see #66 and #78 
- [ ] Update docs for #77 
- [ ] Regenerate data files, version number marked as 0.4.0
- [ ] bump webbpsf version number in setup.py
- [ ] consider bumping DATA_VERSION_MIN in `__init__`, but probably don't. The WFIRST stuff will depend on having that version, but JWST-focused users could keep running with the old data just fine in most cases.

josePhoenix, 2015-11-19:
I'm not sure, but I think this might be the first public version that even tests `DATA_VERSION_MIN`?

mperrin, 2015-11-19:
oh good point. So we might as well push it all the way to 0.4. 

Also
- [ ] write a 'what's new in 0.4' section of the docs.

mperrin, 2015-11-20:
Basically all ready to go, pending any last minute notebook tweaks and possibly #84

mperrin, 2015-11-20:
Looks like the new data file isn't happy on Travis. It's the new NIRISS filter curves which are timing out in the non-pysynphot code path. I remember something like this from making the MIRI updates and I'm sure it's going to be the same fix. 

- [ ] Fix new NIRISS filter files so the interpolation step doesn't time out

mperrin, 2015-11-20:
- [ ] Update NIRCam coronagraph calls to use a version of that code in `webbpsf.optics` instead of the deprecated one in `poppy`

mperrin, 2015-11-20:
- [ ] NIRISS filters fixed and tests, just need to recreate the zip files when I get back to my desktop.

closed 2016-01-21 by mperrin