mperrin/webbpsf #81: Improvements to WebbPSF-WFIRST tutorial notebook

josePhoenix, 2015-11-19:
Based on feedback from @rpvdmarel (and things I noticed when looking with fresh eyes) I have a few things I / we should look at for notebook improvements:

  - [ ] Acknowledge warnings and pink text that users may see, tell them they aren't reasons to mistrust the calculation
  - [ ] In particular, maybe comment out the warning about pysynphot / "local throughput files" for filters, since that's every filter for JWST *and* WFIRST at this point (I think?)
  - [ ] There are certain FAQ items that don't have a good home, but should be discoverable from the notebook (what to do if the kernel dies / what that means, how to restart and clear output)
  - [ ] Explain the panels in the plot shown when `display=True` in the PSF calculation call. Potentially add colorbars. (Does the last panel update with the broadband PSF in the last step? Or is that the last monochromatic step?)
  - [ ] Change the EE example to instead use `display_profiles` and mention the FWHM is marked
  - [ ] remove `GRS_throughput.fits` from the data package and `filters.tsv`
  - [ ] Change explanation of final download link to say 'hover over this link'... potentially expand the explanation by a sentence or two
  - [ ] ensure `image.interpolation` `nearest` is set
  - [ ] Display the final PSF when the calculation completes in the notebook interface (with colorbar, labels, etc)

mperrin, 2015-11-19:
agreed re "maybe comment out the warning"

I think we should avoid writing our own version of a 'Jupyter for new users' basics tutorial, and instead have pointers to existing resources such as http://jupyter-notebook-beginner-guide.readthedocs.org/en/latest/

mperrin, 2015-11-19:
How much of this do you want to get done in the next 24 hrs for 0.4, versus punting to a hypothetical 0.4.1 in mid-December along with the jupyterhub rollout?

josePhoenix, 2015-11-19:
Most, if not all, is pretty straightforward. The only thing I told Roeland was not necessarily straightforward was customizing the appearance of some of the default plots (e.g. adding colorbars to the intermediate optical planes in the `calcPSF(display=True)` call).

josePhoenix, 2015-11-19:
That said, the only thing I think is essential for the release is removing the GRS thing. I think I put that in for testing and never removed it... and it's only going to confuse people.

josePhoenix, 2015-11-20:
FYI, I removed the GRS top hat curve in the webbpsf data sources directory and in filters.tsv. I also doubled the nlambda for the wide filter. (We should come up with less-hand-wavey nlambda at some point, but I can't look at that right now.)

josePhoenix, 2015-11-20:
Closed by #83.

closed 2015-11-20 by josePhoenix