mperrin/webbpsf #85: Orientation of NIRCam LWB mask

kvangorkom, 2015-12-17:
At the request of @mperrin:

The NIRCam LWB mask may be flipped horizontally. WebbPSF currently shows the thinner side of the bar in the science +x direction, while http://adsabs.harvard.edu/abs/2009SPIE.7440E..0WK and correspondence with John S. suggests it ought to be flipped around. The SWB mask appears correct.

mperrin, 2016-01-21:
@kvangorkom can you check with Stansberry to get some of the CV3 coronagraphic image data, and we can double check this against that.

mperrin, 2016-01-26:
For reference: 

[image]

kvangorkom, 2016-01-27:
John S. passed me along CV1 images of the module A and B masks which he'd recently manually transformed into the science frame, and they agree with the image you've attached (I'm not sure whether I should publicly post them here, but I can send them along via email if you like). That said, he was worried about a possible post-DMS flip he hadn't accounted for, so I'll follow up on that and get back to this shortly.

mperrin, 2016-01-27:
What after DMS would introduce an additional flip? The intention is the DMS-output 'Science' frame should be the frame for most scientific analyses, apart from just the rotation of the north angle depending on spacecraft orientation. 

FYI there would be no ITAR concerns with posting the images you mention - "Models and data needed to assess the science performance of the coronagraphs" are explicitly not ITAR and have never been ITAR, per the GSFC export control checklist for JWST.

mperrin, 2016-02-01:
Based on the most recent SIAF draft from Lallo & Cox, I realized we need a 180 deg flip of the entire bar. 

The CV3 image of the occulters posted to the NIRCam team site confirms this. 
(See https://stsci-ins.basecamphq.com/projects/9574738/file/215297942/ModA_MASKRND_SWFPAview.jpeg if you have access to that site)

mperrin, 2016-04-04:
I believe this is all working as desired now.

commit e48a958b (2016-01-28): improve NIRCam coronagraph occulting masks: flip orient of MASKLWB (fixes #85), implement mod A/B layout differences, implement 2x2 ND squares too, and truncate occulter patterns after first sidelobe (as was done for the hardware)
closed 2016-04-04 by mperrin