mperrin/webbpsf #87: SIAF parsing in jwxml may need to be updated ahead of delivery

josePhoenix, 2016-01-12:
Per discussion with Matt Lallo, JWST-STScI-001550 has diverged from reality and we may need to update the parsing code in `jwxml.py`.

 * `DetName` is no longer a valid element to find in a SIAF.
 * `Sci2IdlDegree` is now `Sci2IdlDeg` (Not sure if this change affects the current jwxml implementation?)
 * `UseAfterDate` -- this is already present in our copy of the instrument SIAFs, so it doesn't appear to be a problem

mperrin, 2016-01-12:
Yes, and I ran into a couple other minor issues when trying the lastest pre-delivery SIAF XML a couple days ago. There are some new entries that are null/blank for some fields and that was causing trouble. I agree this will need an update.

josePhoenix, 2016-01-12:
:+1: 

I have been working on a new tool using the jwxml SIAF...

[image]

mperrin, 2016-01-21:
So, the current code seems to be OK to me. I think I already updated the code for these syntax issues at some point in the past. I don't see any reference to `DetName` or `UseAfterDate`, and it already uses `Sci2IdlDeg` instead of `Degree`.  

@josePhoenix have you encountered any specific issues with this while working on your tool development?  For me the code seems to be fine opening the latest SIAF XML files (downloaded from Matt's Box folder).

josePhoenix, 2016-01-21:
I will try again with the file I got from Matt. I had an issue but neglected to save the traceback.

mperrin, 2016-01-21:
I had an issue with the NIRCam aperture names changing, but that's an issue with the XML node text values, not the XML schema itself.  This is fixed in my local code and will make it to github shortly.

mperrin, 2016-01-21:
Oh and Matt had earlier sent me an email asking me to post the following here for him (since he's not on github yet):

From Lallo:

>Very nice work and slick product! This is a great tool for SIAF visualization and QA. 
>
>The latest SIAF specification (reflecting a recent change) can be found in RTC 017 for the "PRDS to Users IRCD, Vol.III" (JWST-STScI-000949).
>
>This JIRA item also tracks the updates to the SIAF Use & Description document JWST-STScI-001550 for consistency with the IRCD:
https://jira.stsci.edu/browse/JWSTSIAF-7
>
>We will be delivering new SIAF content in the new format shortly after the IRCD update is officially approved.
>
>See https://confluence.stsci.edu/display/INSTEL/JWST+SIAF for the latest on SIAF work and access to delivered SIAFs and pre-delivery drafts.

mperrin, 2016-02-01:
Current version of jwxml appears to have no problem reading and displaying the hot-off-the-presses latest NIRCam file. 

@josePhoenix I think we can close this issue - do you see any reason to keep it open? 

[image]

josePhoenix, 2016-02-01:
It must have been something to do with how I was using jwxml in my script... Sorry I haven't had time to provide a proper bug report! This can be closed; I'll make a proper report if I manage to reproduce the issue I was having.

josePhoenix, 2016-02-02:
I see, my issue was fixed in https://github.com/mperrin/webbpsf/commit/220156e91fbf2eb57bc1fc1696a9c5658b524fb5. Closed!

closed 2016-02-02 by josePhoenix