mperrin/webbpsf #88: move poppy.measure_strehl function to webbpsf and ensure it works.

mperrin, 2016-01-21:
See https://github.com/mperrin/poppy/issues/138

The fix is to move that function from ``poppy`` to ``webbpsf``, fixing an oversight from when I fissioned ``poppy`` into its own package separate from ``webbpsf``.

neilzim, 2016-01-28:
I'll also note that the `__init__.py` of webbpsf tries to import measure_strehl from poppy (at line 81), so in addition to moving the function, that statement should be corrected. Because of this, `import webbpsf` fails for the master version.

mperrin, 2016-01-28:
right, I have this fixed in my local branch but hadn't pushed the changes to github yet. I can do that now.

mperrin, 2016-02-08:
Code is working I believe, but we should add a unit test.

mperrin, 2016-02-09:
Unit test added in commit  67373db

closed 2016-02-09 by mperrin