mperrin/webbpsf #90: check handedness of MIRI coronagraph image mask rotations

mperrin, 2016-01-21:
I think I have the sign flipped here. 

Need to ensure consistency with Fig 1. of Bouchet et al. 2015 (http://adsabs.harvard.edu/abs/2015PASP..127..612B). (see attached PNG)

or better yet with actual MIRI output data, properly transformed into the so-called "SCIENCE" frame, e.g. flipped around the way that DMS will output it, which may be different from the MIRI DHAS. 

[image]

@kvangorkom, perhaps you could work with Brendan H. or somebody else on the MIRI team to see if we can get our hands on some CV3 coronagraph images to help check this? I would cc to Laurent and Charles too but I don't think they are on github.

kvangorkom, 2016-01-27:
Dean H. via Brendan H. directed me to [this paper](http://adsabs.harvard.edu/cgi-bin/bib_query?arXiv:1508.02352), which I'm interpreting to suggest that the attached image above shows the correct orientation for the science frame.

Charles L. showed me a MIRI flat field (from CV2, I believe) that agrees with the above and informs me that the raw readout shares the same origin and coordinate convention as the science frame, so it's (maybe) unlikely that we need to worry about real MIRI images adopting different conventions.

mperrin, 2016-01-27:
Cool! I hadn't realized that the DET and SCI frames are the same for MIRI. It appears to be the only instrument for which that is the case! All the H2RG based instruments have coordinate flips between the detector and science frames.   Thanks for checking.

mperrin, 2016-02-08:
After commit f8cb2fe, here's what I'm getting: 

[image]

@kvangorkom can you check that over and confirm if you agree it looks correct now?

kvangorkom, 2016-02-10:
Without doing any more poking around, I'd agree that your image above is consistent with what I'd expect.

closed 2016-02-10 by mperrin