mperrin/webbpsf #100: Testing strategy / changes

josePhoenix, 2016-02-08:
I tried unrolling our various supported combinations into individual builds (https://travis-ci.org/josePhoenix/webbpsf/builds/107336210 https://github.com/josePhoenix/webbpsf/blob/d8662cd6b2251f1f5abf99b331168220343082e2/.travis.yml ) but a) it takes too long to run every combination (which is what I was trying to evaluate the feasibility of, so fair enough) and b) there is no conda package for astropy matching some of these permutations, causing the corresponding build to fail.

In theory we support the cartesian product of these sets of environment states:

* Python
  * 2.7.x
  * 3.5.x
* NumPy
  * 1.8
  * 1.9
  * 1.10
* Astropy
  * development version
  * stable versions > 1.0.1
* POPPY
  * development version
  * latest stable version
* PySynphot
  * present
  * absent

In practice, I think we can only realistically test some of these. There is no conda package built against NumPy 1.8 for Astropy > 1.0.1 (see https://travis-ci.org/josePhoenix/webbpsf/jobs/107336226). POPPY evolves in parallel with WebbPSF, so tests against the last released version are prone to failure. I haven't investigated the cause of this test failure, but it looks pretty gnarly: https://travis-ci.org/josePhoenix/webbpsf/jobs/107336241.

I guess my question is, are we okay supporting configurations we don't test in? We're currently testing this subset of configurations:

  * Python 2.7, NumPy 1.9, Astropy dev, POPPY dev
  * Python 2.7, NumPy 1.9, Astropy "stable" (latest packaged for conda with this NumPy, 1.0.4), POPPY dev
  * Python 3.4, NumPy 1.9, Astropy "stable", POPPY dev
  * Python 2.7, NumPy 1.8, Astropy "stable" (in practice 0.4.1!?), POPPY dev
  * Python 2.7, NumPy 1.9, Astropy "stable", POPPY stable (from PyPI)
  * Python 2.7, NumPy 1.8, Astropy "stable" (0.4.1 again), POPPY stable

I propose a new test matrix of:

  * Python
    * 2.7.x
    * 3.5.x
  * Numpy
    * 1.9
    * 1.10
  * Only stable Astropy (unless we encounter another weird packaging issue)
  * POPPY
    * dev version
    * stable (but put that build under `allow_failures` to cut down on noise)

That's eight builds. There's a question of if we want to test in the presence/absence of PySynphot, in which case we need to either double that (to 16) or go through and isolate PySynphot-dependent code with some configurable switch. (This might also involve pushing that code down to POPPY.)

mperrin, 2016-02-08:
I'm certainly on board with simplifying things as much as possible. 
Certainly I'm on board with dropping numpy 1.8; I see no reason for that given recent Conda and Ureka are on at least 1.9.  
 
I don't see a strict need to necessarily test every possible combination of the matrix's axes. For instance we could test dev poppy on numpy 1.10 but not numpy 1.9.  I suspect that many classes of potential failures are to some extent independent of some of these axes; for instance I think it's unlikely that we'd have a case that fails with dev poppy and Python 3.5 and numpy 1.10, and yet doesn't also fail with dev poppy and Py 3.5 and numpy 1.9. 

An alternate approach to picking test cases might be to pick a single "master" test case (py3.5, numpy 1.9, stable astropy, dev poppy?) and then add several other test cases that each differ on a single one of those axes. It's less comprehensive but I suspect would still do a good job in catching failures. 
That might give us as few as 5 test cases needed:

- Python 3.5, Numpy 1.9, stable astropy, dev poppy, no pysynphot
- Python 2.7, Numpy 1.9, stable astropy, dev poppy, no pysynphot
- Python 3.5, Numpy 1.10, stable astropy, dev poppy, no pysynphot
- Python 3.5, Numpy 1.9, stable astropy, stable poppy, no pysynphot
- Python 3.5, Numpy 1.9, stable astropy, dev poppy, with pysynphot

It's less comprehensive but I certainly don't think that pysynphot makes enough of a difference that we need to duplicate the entire test suite in all permutations to cover it. There's only a handful of functions where that code gets exercised, and the majority of the code base here is totally independent of whether that's installed or not.

mperrin, 2016-06-10:
Partially addressed in 0.5.0, could still use some attention perhaps?  Particularly w.r.t the Travis py 3.5 issues.

mperrin, 2016-12-05:
Current Travis setup has been updated and seems to be working well. I think this issue can be closed now. @josePhoenix do you agree?

mperrin, 2017-08-11:
stale issue. dealt with already; no action needed now.

closed 2017-08-11 by mperrin