mperrin/webbpsf #102: Update min/max wavelengths for SIs, particularly NIRCam

mperrin, 2016-02-29:
From Jarron Leisenring:

> The first [issue] deals with NIRCam's minimum and maximum wavelengths. WebbPSF now invokes hard cut-offs at [0.6, 2.35] microns for the SW and [2.35, 5.0] for the LW channels. However, the filter profiles (ie., http://www.stsci.edu/jwst/instruments/nircam/instrumentdesign/filters/nircam_passbands_large.jpg) show sensitivities are slightly beyond these wavelengths. I would suggest allowing some buffer room on either side:
>   SW - [0.5, 2.5]
>   LW - [2.3, 5.3]
> You may also want to consider this for NIRISS as well. I'm calculating monochromatic wavelengths while stepping through filter curves to simulate grism spectroscopy, so going to the wings of the filter profiles would be useful.

mperrin, 2016-02-29:
Making `LONG_WAVELENGTH_MAX=5.3e-6` for NIRCam (and NIRISS, NIRSpec too, likely) is straightforward. 
Having overlapping cutoffs between the SW and LW is a little trickier to implement given the existing code in `NIRCam._validateConfig`. One possible fix might be to check the *average* wavelength of a given calculation and set the bandpass based on that. So e.g. if the average wavelength of F150W2 is < 2.35 (which it is of course) then it would be OK for some individual wavelengths to be longer than that cutoff. 

Hmm, though that doesn't quite address Jarron's use case of calling with the monochromatic option, which would automatically flip over at whatever the cutoff wavelength is.  I don't see any non-magical way of making the API guess whether a `calcPSF(monochromatic=2.35e-6)` call is supposed to be the very long end of F150W2 or the very short end of F277W. The cleanest solution might be to require such use cases to explicitly set the pixel scale?

mperrin, 2016-03-02:
I talked with John Stansberry today and he confirmed the dichroic beam splitter's cutoff is indeed sufficiently gradual that there is this minor overlap at edge wavelengths.

JarronL, 2016-03-02:
I did not fully appreciate the issue with having `SHORT_WAVELENGTH_MAX` greater than `LONG_WAVELENGTH_MAX`. Keeping `SHORT_WAVELENGTH_MAX = LONG_WAVELENGTH_MIN = 2.35e-6` probably makes the most sense. This is a sensible wavelength cut-off for the LW filters and the use-case for SW observations is extremely small, specifically spectral dispersion of F150W2 using DHS. Even for that case, the longer wavelengths of the spectrum get contaminated by the 2nd order dispersion, so precise monochromatic simulations are not critical past 2.35 um.

mperrin, 2016-03-03:
OK, we can keep the short/long cutoff as is, but increase the long wavelength cutoff to 5.3 µm. 

For your specific case of explicitly wanting simulations > 2.35 µm in short wave, you should be able to easily work around that by explicitly setting the `pixelscale` attribute before the calculation. If you set the `pixelscale` to any non-default value that's sufficient to disable the auto-switching between SW and LW. This is true even if you do something like  e.g. `nircam.pixelscale = nircam._pixelscale_short*1.000001`

closed 2016-04-04 by mperrin