mperrin/webbpsf #103: Tk GUI bug with matplotlib backend MacOSX

mperrin, 2016-02-29:
**Jarron Leisenring reported the following:** 

My other problem appears to be a comparability issue involving tkinter while trying to start the gui. The full error is below. I discovered that this can be mitigated by setting matplotlib.use('TkAgg') immediately after importing matplotlib. By default the matplotlib backend is 'MacOSX'. I also appended my webbpsf.system_diagnostic report.

OS: Darwin-11.4.2-x86_64-i386-64bit
Python version: 2.7.11 |Anaconda 2.1.0 (x86_64)| (default, Dec  6 2015, 18:57:58)  [GCC 4.2.1 (Apple Inc. build 5577)]
numpy version: 1.10.4
poppy version: 0.4.0
webbpsf version: 0.4.0

tkinter version: 0.3.1
wxpython version: not found

astropy version: 1.1.1
pysynphot version: 0.9.7
pyFFTW version: 0.10.1

Floating point type information for numpy.float:
Machine parameters for float64
---------------------------------------------------------------
precision= 15   resolution= 1.0000000000000001e-15
machep=   -52   eps=        2.2204460492503131e-16
negep =   -53   epsneg=     1.1102230246251565e-16
minexp= -1022   tiny=       2.2250738585072014e-308
maxexp=  1024   max=        1.7976931348623157e+308
nexp  =    11   min=        -max
---------------------------------------------------------------

Floating point type information for numpy.complex:
Machine parameters for float64
---------------------------------------------------------------
precision= 15   resolution= 1.0000000000000001e-15
machep=   -52   eps=        2.2204460492503131e-16
negep =   -53   epsneg=     1.1102230246251565e-16
minexp= -1022   tiny=       2.2250738585072014e-308
maxexp=  1024   max=        1.7976931348623157e+308
nexp  =    11   min=        -max
---------------------------------------------------------------

In [8]: webbpsf.gui()
2016-02-29 12:31:20.498 python[94726:60b] -[NSApplication _setup:]: unrecognized selector sent to instance 0x10857fe40
2016-02-29 12:31:20.500 python[94726:60b] An uncaught exception was raised
2016-02-29 12:31:20.500 python[94726:60b] -[NSApplication _setup:]: unrecognized selector sent to instance 0x10857fe40
2016-02-29 12:31:20.505 python[94726:60b] (
   0   CoreFoundation 0x00007fff8db53f56 __exceptionPreprocess + 198
   1   libobjc.A.dylib 0x00007fff8113cd5e objc_exception_throw + 43
   2   CoreFoundation 0x00007fff8dbe01be -[NSObject doesNotRecognizeSelector:] + 190
   3   CoreFoundation 0x00007fff8db40e23 ___forwarding___ + 371
   4   CoreFoundation 0x00007fff8db40c38 _CF_forwarding_prep_0 + 232
   5   libtk8.5.dylib 0x000000011132676d TkpInit + 472
   6   libtk8.5.dylib 0x000000011129ef91 Initialize + 1541
   7   _tkinter.so 0x0000000110ee6a3d Tcl_AppInit + 77
   8   _tkinter.so 0x0000000110ee4485 Tkinter_Create + 997
   9   libpython2.7.dylib 0x00000001000c74bc PyEval_EvalFrameEx + 40124
   10  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   11  libpython2.7.dylib 0x000000010003e100 function_call + 176
   12  libpython2.7.dylib 0x000000010000c552 PyObject_Call + 98
   13  libpython2.7.dylib 0x000000010001dbed instancemethod_call + 333
   14  libpython2.7.dylib 0x000000010000c552 PyObject_Call + 98
   15  libpython2.7.dylib 0x00000001000bc547 PyEval_CallObjectWithKeywords + 87
   16  libpython2.7.dylib 0x000000010002211e PyInstance_New + 126
   17  libpython2.7.dylib 0x000000010000c552 PyObject_Call + 98
   18  libpython2.7.dylib 0x00000001000c0013 PyEval_EvalFrameEx + 10259
   19  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   20  libpython2.7.dylib 0x00000001000c567d PyEval_EvalFrameEx + 32381
   21  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   22  libpython2.7.dylib 0x000000010003e100 function_call + 176
   23  libpython2.7.dylib 0x000000010000c552 PyObject_Call + 98
   24  libpython2.7.dylib 0x000000010001dbed instancemethod_call + 333
   25  libpython2.7.dylib 0x000000010000c552 PyObject_Call + 98
   26  libpython2.7.dylib 0x00000001000792d8 slot_tp_init + 88
   27  libpython2.7.dylib 0x0000000100076155 type_call + 245
   28  libpython2.7.dylib 0x000000010000c552 PyObject_Call + 98
   29  libpython2.7.dylib 0x00000001000c0013 PyEval_EvalFrameEx + 10259
   30  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   31  libpython2.7.dylib 0x00000001000c567d PyEval_EvalFrameEx + 32381
   32  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   33  libpython2.7.dylib 0x00000001000c567d PyEval_EvalFrameEx + 32381
   34  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   35  libpython2.7.dylib 0x00000001000c6238 PyEval_EvalFrameEx + 35384
   36  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   37  libpython2.7.dylib 0x00000001000c567d PyEval_EvalFrameEx + 32381
   38  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   39  libpython2.7.dylib 0x00000001000c567d PyEval_EvalFrameEx + 32381
   40  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   41  libpython2.7.dylib 0x00000001000c567d PyEval_EvalFrameEx + 32381
   42  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   43  libpython2.7.dylib 0x00000001000c567d PyEval_EvalFrameEx + 32381
   44  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   45  libpython2.7.dylib 0x00000001000c567d PyEval_EvalFrameEx + 32381
   46  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   47  libpython2.7.dylib 0x00000001000c567d PyEval_EvalFrameEx + 32381
   48  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   49  libpython2.7.dylib 0x000000010003e100 function_call + 176
   50  libpython2.7.dylib 0x000000010000c552 PyObject_Call + 98
   51  libpython2.7.dylib 0x00000001000c2538 PyEval_EvalFrameEx + 19768
   52  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   53  libpython2.7.dylib 0x00000001000c567d PyEval_EvalFrameEx + 32381
   54  libpython2.7.dylib 0x00000001000c83a3 PyEval_EvalCodeEx + 2131
   55  libpython2.7.dylib 0x00000001000c84c6 PyEval_EvalCode + 54
   56  libpython2.7.dylib 0x00000001000ede5e PyRun_FileExFlags + 174
   57  libpython2.7.dylib 0x00000001000ee0fa PyRun_SimpleFileExFlags + 458
   58  libpython2.7.dylib 0x000000010010517d Py_Main + 3165
   59  python 0x0000000100000f54 start + 52
   60  ??? 0x0000000000000003 0x0 + 3
)
2016-02-29 12:31:20.506 python[94726:60b] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplication _setup:]: unrecognized selector sent to instance 0x10857fe40'
terminate called throwing an exception/Users/jwstnircam/anaconda/envs/python2/bin/python.app: line 3: 94726 Abort trap: 6 /Users/jwstnircam/anaconda/envs/python2/python.app/Contents/MacOS/python "$@"

josePhoenix, 2016-03-01:
I've reproduced the bug and found a workaround. Basically, when using the Tk GUI, matplotlib must be configured with the TkAgg backend *before* the pyplot helper is imported. See http://stackoverflow.com/a/34109240 for details.

The workaround is to edit `__init__.py` and add

```
import matplotlib
matplotlib.use('TkAgg')
```

just before the line `from . import utils`. The downside is that (I believe) this will break notebook use of WebbPSF, so it's not a complete solution on its own. I think solving this completely will require eliminating use of the pyplot API to matplotlib in WebbPSF and POPPY, as importing it has the side-effect of setting the matplotlib backend.

mperrin, 2016-03-01:
well we certainly don't want to break the notebook!  I think for now we should add documentation about setting the backend appropriately, and maybe leave it at that.  Converting everything away from pyplot to pure matplotlib object API sounds like a lot of work for minimal reward.

mperrin, 2018-08-28:
Issues migrated to https://github.com/spacetelescope/webbpsf/issues. 
Please see the copies in that repo from now on.

closed 2018-08-28 by mperrin