mperrin/webbpsf #108: Update data files on Central Storage

pllim, 2016-04-27:
When I set `WEBBPSF_PATH` to the value that I am supposed to use as STScI user (http://pythonhosted.org/webbpsf/installation.html#data-install) in the SSBDEV environment, I get this error:
```python
EnvironmentError: WebbPSF data package has version 0.4.0, but 0.4.1 is needed. See http://pythonhosted.org/webbpsf/installation.html#data-install for a link to the latest version.
```

This implies that data files need to be updated on Central Storage for STScI users.

c/c @bhilbert4

josePhoenix, 2016-04-27:
Thanks for the report. I'm updating them now.

pllim, 2016-04-27:
Somewhat related, SSBX is not longer working nor supported. So, it should be removed from STScI instructions. See http://ssb.stsci.edu/ssb_software.shtml for more info.

josePhoenix, 2016-04-27:
I wasn't aware that had been officially released. Internal users can also install WebbPSF from there, and they will (should?) get the corresponding data package automatically.

josePhoenix, 2016-04-27:
Okay, the Central Storage files have been updated. `/grp/jwst/ote/webbpsf-data` is a link to `/grp/jwst/ote/webbpsf-data-0.4.1` and either path should work.

pllim, 2016-04-27:
@josePhoenix , since SSBX is not working, most JWST users will be switching to SSBDEV, which picks up the dev version of WebbPSF. I think that is the source of this error.

If you want to freeze WebbPSF version on SSBDEV, maybe we can bring @jhunkeler into that conversation, if it is even possible.

mperrin, 2016-04-27:
Whoa, the Conda distribution is now live too?  I did not know that. 

This seems worthy of wider announcement to the science staff.   

In particular, when I log in to witservN or telservN, ssbx is still my default environment and I can't recall how to change that to default to ssbdev instead. Brief migration instructions might be appreciated by many, I suspect.

pllim, 2016-04-27:
Re: Conda, let's discuss that in #109

pllim, 2016-04-27:
Thanks, @josePhoenix !

closed 2016-04-27 by pllim

mperrin, 2016-04-27:
No the source of this error is that we actually did a new release recently and bumped the version number and I forgot to update the central store files. So this is distinct from using the dev copy. 

Users of SSBDEV will need to be aware that webbpsf may break randomly at times in the same manner as any other dev package, but presumably that's no more or less a concern than anything else in the nightly builds.

pllim, 2016-04-27:
@mperrin -- hopefully the transition to Conda will solve this problem of relying on SSBDEV.