mperrin/webbpsf #109: AstroConda released, update installation instructions accordingly

josePhoenix, 2016-04-27:
Per @pllim, we can now tell users about the `conda` packages @jhunkeler made

josePhoenix, 2016-04-27:
http://astroconda.readthedocs.org/en/latest/installation.html

pllim, 2016-04-27:
Should wait for confirmation from @jhunkeler for this one. I don't think JWST stuff is moved over yet.

josePhoenix, 2016-04-27:
You brought up a good point in the other issue. I hadn't thought about how SSBdev interacts with our data package versioning... we don't have "nightly builds" for the instrument reference data, so it's possible that the (relatively infrequent) changes involving will break WebbPSF nightlies.

mperrin, 2016-04-27:
I also want to understand more about the handling of the data packages - I know Conda is user selectable for what gets installed, but is the _default_ behavior going to be for all users to get our 400 MB of reference files?  That may not make the most sense.

pllim, 2016-04-27:
@jhunkeler -- Please clarify. Thanks.

mperrin, 2016-04-27:
@josePhoenix anybody who wants/requires stability should be using the REL versions not DEV. 

My understanding from reading http://astroconda.readthedocs.org/en/latest/faq.html is that AstroConda will function more or less like REL, and is therefore a good choice for most users?

josePhoenix, 2016-04-27:
Based on the [manifest](http://ssb.stsci.edu/astroconda/osx-64/) these packages are installed by default when following the recommended installation instructions for AstroConda:

```
astropy >=1.1
cfitsio ==3.370
d2to1 ==0.2.12
ds9 ==7.4
fftw ==3.3.4
htc_utils >=0.1
imexam ==0.5.2
numpy 1.10*
photutils ==0.2.1
poppy ==0.4.0
purge_path >=0.0.1
pyds9 ==1.8.1
pyfftw ==0.9.2
python 3.4*
stsci-data-analysis
stsci-hst
webbpsf ==0.4.0
webbpsf-data ==0.4.0
```

I would say let them install `webbpsf-data` until/unless we hear complaints about it. People installing Ureka generally wanted things to "just work", and I suspect these people will be using the `stsci` metapackage to bring in everything and the kitchen sink.

josePhoenix, 2016-04-27:
However, I just noticed it hasn't picked up 0.4.1 yet... another thing to ask @jhunkeler

pllim, 2016-04-27:
Most STScI users who use WebbPSF *and* JWST pipeline will naturally use DEV. So, it is best to make sure it works both in REL and DEV reasonably well.

pllim, 2016-04-27:
Re: Data files -- It is a philosophy issue. There is no right answer.

josePhoenix, 2016-04-27:
I would suggest a `/grp/jwst/ote/webbpsf-data-dev` folder to be updated from our stash of source material when we make changes to the data. Then the `ssbdev` environment could include an `export` line pointing the `WEBBPSF_PATH` there. This would also improve cases like @mperrin needing my IFU PSF reference data to evaluate a pull request. The downside is we'll forget to update it, since it's far too slow to use over NFS in day-to-day dev work.

mperrin, 2016-04-27:
"So, it is best to make sure it works both in REL and DEV reasonably well."   

Agreed. We don't plan to leave it broken in DEV for any great length of time. This will probably just make us (me in particular?) get better about doing development not in master. 

Relatedly now that you added me to the space telescope github group, I was wondering if I should move the main webbpsf repository there instead of in my own account.

josePhoenix, 2016-04-27:
(+1 to putting it under SpaceTelescope. If you do, be sure to make a new empty repository and not 'fork' it from here.)

jhunkeler, 2016-04-27:
* 0.4.1 hasn't been distributed to the public repo yet, but I confirm it builds/works perfectly fine (the data files too), so I'm not worried about that. 

* Yes, the entire dataset is installed with webbpsf by default, and the WEBBPSF_PATH variable is automatically defined to point to `/conda/env/path/share/webbpsf-data`.

* We received the "green light" yesterday to make the release public, so despite the lack of an official announcement at the moment, you can go ahead and use it and "tell your friends". That's not a problem.

pllim, 2016-04-27:
Do you use Travis CI religiously? If so, I think @sosey said that there are too many repo on Space Telescope or something, and that can mess with Travis CI jobs being queued up too long.

mperrin, 2016-04-27:
Hmm, yes. I have tests for webbpsf & poppy both running in my own Travis account.

pllim, 2016-04-27:
In addition, when you move the repo, your forks from this one are going to disappear. Something to think about.

josePhoenix, 2016-04-27:
@jhunkeler how should we ensure new releases get packaged and distributed through AstroConda? Should we add "email jhunkeler" to the release checklist after we tag it?

pllim, 2016-04-27:
@nmearl, did you have problem with forks when you moved `specviz` to Space Telescope?

sosey, 2016-04-27:
The problem with running CI on Travis is there's a limit to the number of tests an organization can be running at one time. We've hung travis in the past and had to kill off processes to get the tests to complete. My recommendation before was that people run tests on their private repos while they are developing to lessen the chance of overload. Also, remember to check occasionally that your tests aren't hung and blocking people, especially with the much larger number of repos we now have.

jhunkeler, 2016-04-27:
@josePhoenix fork https://github.com/astroconda/astroconda-contrib and make a pull request against webbpsf, webbpsf-data, and/or poppy. On my end it's just a simple merge and compilation.

mperrin, 2016-04-27:
OK, let's hold off on any repo relocations for now. Too many potential complications with Travis and breaking existing fork links which I don't want to deal with, especially at the same time as the initial release of astroconda. Let's only change one big thing at a time. :-)

pllim, 2016-04-27:
Re: Moving repo -- You can always open an issue about it and see if anyone complains :smile:

josePhoenix, 2016-04-27:
This would be how to do it, but I think concerns about limited Travis resources might be reason enough not to. https://help.github.com/articles/transferring-a-repository/#transferring-from-a-user-to-an-organization

josePhoenix, 2016-08-22:
Fixed in #126

closed 2016-08-22 by josePhoenix