mperrin/webbpsf #117: update/replace precomputed PSFs linked from web page

mperrin, 2016-05-24:
These should be replaced once the next version of webbpsf is available. 
In particular not just the updated WFE info - we also want to remove the two vestigial files for filters that haven't been included in NIRCam for many years.

mperrin, 2016-06-10:
@josePhoenix is this something you could perhaps help with next week too please? Updating the library of precomputed example PSF FITS files at http://www.stsci.edu/~mperrin/software/psf_library/
You could add some similar files for WFI as well.

Should be a pretty trivial bit of scripting to make a new library of PSF files. It would probably be faster to write a new for loop script to generate them than for me to try to find the script I used three years ago.  If you could do that and put the files somewhere on Central Store then I can copy them to the right directory for the web server. 

If you're feeling ambitious, that whole page needs to eventually get moved from my personal home directory to a page in the new Jahia CMS (under https://jwst.stsci.edu/science-planning/performance--simulation-tools-1/psf-simulation-tool-webbpsf, which could also use some text updates).  You could coordinate that with Sherie H. who is supposed to be getting trained on how to use Jahia. But the training might not have happened yet so this may need to wait until later.

josePhoenix, 2016-09-07:
Quick update on this issue: I have a PSF library maker in https://github.com/josePhoenix/webbpsf/blob/psf-library-maker/dev_utils/compute_psf_library.py (josePhoenix/webbpsf@psf-library-maker). I've gotten access to the Jahia CMS, too.

Are we waiting on some WFE model updates, or should I press ahead with this @mperrin ?

mperrin, 2016-09-07:
Waiting on WFE model updates, yes. Stay tuned.

mperrin, 2018-07-09:
The precomputed PSF library is deprecated, and we shouldn't do any further work on this.

closed 2018-07-09 by mperrin