mperrin/webbpsf #122: Outdated attributes used in examples on documentation site

kjbrooks, 2016-06-08:
While working through some of the WebbPSF examples, I found that several of them were calling `webbpsf.calc_or_load_psf` which didn't work and doesn't seem to exist anymore. 

I found three instances of this on [the examples page](https://pythonhosted.org/webbpsf/more_examples.html)

mperrin, 2016-06-08:
I can certainly update all those. At this point though I wonder about the usefulness of having a page with example code as part of the HTML documentation, as opposed to just putting those all in a Jupyter notebook and linking to it via nbviewer. The latter would be easier to maintain long term. What do you think?

josePhoenix, 2016-06-08:
If we were feeling ambitious, we could compile our notebooks into our docs directly: https://github.com/ipython/nbconvert/pull/35 https://github.com/ipython/ipython/issues/4936

mperrin, 2016-06-08:
Oh that looks amazing. This seems to be the best reference:  http://nbsphinx.readthedocs.io/en/0.2.8/index.html

I tried to get this running in a local branch. First of course I run into an issue that some of the astropy helper sphinx extensions conflict with this - specifically the "autosummary" isn't smart enough to ignore notebook files.  So I tried just deleting all extensions entirely. Now I'm getting:
```
NotJSONError: Notebook does not appear to be JSON: u'{\n "cells": [\n  {\n   "cell_type": "...
```
which makes me suspect a version incompatibility, but I'm running Jupiter 4.1.0 which is pretty recent. 
Sigh. Nothing is ever easy

closed 2016-06-10 by mperrin