mperrin/webbpsf #127: conda install webbpsf followed by conda install notebook downgrades WebbPSF

josePhoenix, 2016-07-11:
It appears that AstroConda does not provide a WebbPSF 0.5 package that is compatible with the version of the Jupyter Notebook available from Continuum Analytics. It's not clear whether this should be fixed by our Science Software Branch with a package stating compatibility with NumPy 1.10 (cc @jhunkeler) or by Continuum Analytics rebuilding the `notebook` package with the latest NumPy.

¯¯\\_(ツ)_/¯¯

To reproduce:

1. `conda create -n somename stsci` - install main AstroConda metapackage
2. Verify WebbPSF 0.5.0 is installed
3. `conda install notebook` - install the Jupyter Notebook (packaged by Continuum)
4. Verify that WebbPSF 0.4.0 is now the installed version

Full log of replication below:

```
00:28:53 tanuki:~ jlong$ conda create -n testcasewebbpsfdowngrade stsci
Using Anaconda Cloud api site https://api.anaconda.org
Fetching package metadata: ........
Solving package specifications: .........

Package plan for installation in environment /Users/jlong/anaconda/envs/testcasewebbpsfdowngrade:

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    pandas-0.18.1              |      np111py35_0         7.1 MB
    pyfftw-0.9.2               |      np111py35_1          88 KB
    scikit-learn-0.17.1        |      np111py35_2         3.6 MB
    stsci.ndimage-0.10.1       |      np111py35_0         149 KB
    stsci.tools-3.4.1          |      np111py35_0         411 KB
    calcos-3.1.8               |      np111py35_0         368 KB
    fitsblender-0.2.6          |      np111py35_0          61 KB
    nictools-1.1.3             |      np111py35_0         109 KB
    pyqtgraph-0.9.11           |      np111py35_1         934 KB
    stginga-0.1.1              |      np111py35_0          87 KB
    stistools-1.1              |      np111py35_0          68 KB
    stsci.image-2.2.0          |      np111py35_0          22 KB
    stsci.imagemanip-1.1.2     |      np111py35_0          10 KB
    stsci.numdisplay-1.6.1     |      np111py35_0          41 KB
    stwcs-1.2.3                |      np111py35_0         176 KB
    wfpc2tools-1.0.3           |      np111py35_0          30 KB
    astroimtools-0.1           |      np111py35_0          35 KB
    costools-1.2.1             |      np111py35_0          40 KB
    imexam-0.5.2               |      np111py35_0         335 KB
    poppy-0.5.0                |      np111py35_0         232 KB
    pysynphot-0.9.8.3          |      np111py35_0         4.4 MB
    scikit-image-0.12.3        |      np111py35_1        17.8 MB
    specview-0.1               |      np111py35_0         134 KB
    stsci.sphere-0.2           |      np111py35_1         333 KB
    wfc3tools-1.3.1            |      np111py35_0         3.5 MB
    acstools-2.0.2             |      np111py35_0          55 KB
    photutils-0.2.1            |      np111py35_1         256 KB
    reftools-1.7.1             |      np111py35_1         113 KB
    webbpsf-0.5.0              |      np111py35_0         169 KB
    drizzlepac-2.1.4           |      np111py35_0         716 KB
    cube-tools-0.0.0           |      np111py35_0          59 KB
    asv-0.1.1                  |      np111py35_0         857 KB
    stsci-data-analysis-1.0.0  |      np111py35_2          951 B
    stsci-hst-1.0.3            |      np111py35_0           1 KB
    stsci-1.0.0                |      np111py35_1           1 KB
    ------------------------------------------------------------
                                           Total:        42.1 MB

The following NEW packages will be INSTALLED:

    acstools:            2.0.2-np111py35_0
    alabaster:           0.7.8-py35_0
    appdirs:             1.4.0-py35_0
    appnope:             0.1.0-py35_0
    asdf:                1.0.3-np111py35_0
[... 261 lines cut ...]
    reftools:            1.7.1-np111py35_1              --> 1.7.1-np110py35_0
    stginga:             0.1.1-np111py35_0              --> 0.0.0-np110py35_0
    stsci:               1.0.0-np111py35_1              --> 1.0.0-np110py35_0
    stsci-data-analysis: 1.0.0-np111py35_2              --> 1.0.0-np110py35_0
    stsci-hst:           1.0.3-np111py35_0              --> 1.0.1-np110py35_0
    stsci.sphere:        0.2-np111py35_1                --> 0.2-np110py35_0
    webbpsf:             0.5.0-np111py35_0              --> 0.4.0-np110py35_0
    webbpsf-data:        0.5.0-0                        --> 0.4.0-0

Proceed ([y]/n)?

Fetching packages ...
python-3.5.2-0 100% |####################################################################| Time: 0:00:01  11.32 MB/s
scipy-0.17.1-n 100% |####################################################################| Time: 0:00:01   9.04 MB/s
pyfftw-0.9.2-n 100% |####################################################################| Time: 0:00:00   1.04 MB/s
scikit-learn-0 100% |####################################################################| Time: 0:00:00   5.62 MB/s
scikit-image-0 100% |####################################################################| Time: 0:00:01  10.46 MB/s
Extracting packages ...
[      COMPLETE      ]|#######################################################################################| 100%
Unlinking packages ...
[      COMPLETE      ]|#######################################################################################| 100%
Linking packages ...
[      COMPLETE      ]|#######################################################################################| 100%
(testcasewebbpsfdowngrade)00:32:26 tanuki:~ jlong$
```

mperrin, 2016-07-11:
I don't understand - what requirements are forcing this downgrade? 

I think it's not anything in `webbpsf` or `poppy` themselves, but has to be something in the `astroconda-contrib` specifications. The requirements in poppy are just  'numpy>=1.8.0' so any version above that should be OK. Right?  But conversely I don't see what's wrong with the `meta.yaml` files for either in astroconda.

josePhoenix, 2016-07-11:
I think it's even more ridiculous. I think it might be that Continuum Analytics has not built a `notebook` package with the appropriate magic bits that say it works with NumPy 1.11. @jhunkeler and I talked it over and iirc that's what we came up with.

josePhoenix, 2016-07-11:
Conversely, I suppose, we have not built a webbpsf/webbpsf-data/poppy package set for the new version that says it  works with 1.10. That should be within our capabilities to fix.

mperrin, 2016-07-11:
But that's what confuses me.  I don't see anything in https://github.com/astroconda/astroconda-contrib/blob/master/poppy/meta.yaml which indicates that it's not compatible with numpy 1.10. 
Same for https://github.com/astroconda/astroconda-contrib/blob/master/webbpsf/meta.yaml

jhunkeler, 2016-07-11:
The blame is split between Continuum and myself. I stopped building against numpy 1.10 in favor of 1.11 to cut down on the overall build-time. It is Continuum's fault for simply not keeping their packages up to date with the latest numpy.

Why did I go from 1.10 to 1.11 so suddenly? At the time, Continuum compiled Astropy 1.2.1 against numpy 1.11, but not 1.10 and some of our software already relied on features in Astropy 1.2.1. 

Tough situation -- poor execution, I'll admit.

In any case, I posted a message on the Anaconda mailing list when this happened. It took a few days (a few days too long), but they ended up recompiling Astropy 1.2.1 against numpy >=1.8. We should probably just tell them about `notebook`... there's high probability they have no idea it's even missing 1.11 support.

josePhoenix, 2016-07-12:
Thanks @jhunkeler ! Just built a new astroconda env with `conda create --name astroconda python=3.5 stsci notebook` and successfully imported WebbPSF v0.5.0.

closed 2016-07-12 by josePhoenix