mperrin/webbpsf #129: Replace PySynphot dependency in setup.py with one that references the repo

josePhoenix, 2016-07-26:
Looks like this is possible: http://stackoverflow.com/questions/18026980/python-setuptools-how-can-i-list-a-private-repository-under-install-requires

josePhoenix, 2016-10-12:
Upon investigation, this workaround is not in fact possible. It was deprecated, and now it doesn't work at all. I guess we should plan ahead for the day where we have to drop PySynphot.

mperrin, 2016-10-12:
I don't understand - isn't pysynphot still used as one of the low level infrastructure pieces in Pandeia? Why would we have to drop it?

josePhoenix, 2016-10-12:
Because we want users to be able to install WebbPSF and POPPY using the standard Python package manager, but SSB does not want to maintain PyPI packages for pysynphot.

mperrin, 2016-10-12:
Right. But it's already an optional dependency, so users can either go without it (and still be able to use WebbPSF just as now), or install Astroconda and get pysynphot that way. As far as I know it's going to remain available in astroconda even if not on PYPI. So I don't see a need to drop the existing functionality given that. I agree this is less than ideal though.

mperrin, 2017-08-11:
no action needed on this at this time, especially since increasingly astroconda is the main distribution channel for all this .

closed 2017-08-11 by mperrin