mperrin/webbpsf #130: Cannot build for Python 3.5.2

jhunkeler, 2016-07-29:
## Failure message
```
BUILD START: webbpsf-1.2.3.dev2-np111py35_0
Fetching package metadata: ........
Solving package specifications: ...................................................................Initialized empty Git repository in /srv/iraf/bldtmp/porcelain.ekN6011skD/porcelain/conda-bld/git_cache/__github.com_mperrin_webbpsf/
Initialized empty Git repository in /srv/iraf/bldtmp/porcelain.ekN6011skD/porcelain/conda-bld/work/.git/
Submodule 'astropy_helpers' (https://github.com/astropy/astropy-helpers.git) registered for path 'astropy_helpers'
Submodule 'webbpsf/jwxml' (git://github.com/mperrin/jwxml.git) registered for path 'webbpsf/jwxml'
Initialized empty Git repository in /srv/iraf/bldtmp/porcelain.ekN6011skD/porcelain/conda-bld/work/astropy_helpers/.git/
Submodule path 'astropy_helpers': checked out '90de2d8340d1eafaface009625056ca62f5a6c76'
Initialized empty Git repository in /srv/iraf/bldtmp/porcelain.ekN6011skD/porcelain/conda-bld/work/webbpsf/jwxml/.git/
Submodule path 'webbpsf/jwxml': checked out '1b8795c36f216d70392164389d2491c09b80748f'
+ python setup.py install
Traceback (most recent call last):
  File "setup.py", line 38, in <module>
    conf = config.ConfigParser()
AttributeError: module 'distutils.config' has no attribute 'ConfigParser'
+ exit 1
```

The Python devs replaced `distutils.config.ConfigParser` with `distutils.config.RawConfigParser` in the latest release of Python 3k. Also, I'm pretty sure this bug is being handled properly in the latest release of the astropy packaging template by deprecating the use of `distutils.config` for configuration management altogether.

## Example fixes

https://github.com/astropy/photutils/blob/master/setup.py#L24
https://github.com/spacetelescope/gwcs/blob/master/setup.py#L24

josePhoenix, 2016-07-29:
Another example fix: https://github.com/mperrin/webbpsf/pull/128 😉 

I'll go ahead and merge it myself.

josePhoenix, 2016-07-29:
@jhunkeler master should build now. Let me know if it succeeds and I'll close this issue.

jhunkeler, 2016-08-01:
@josePhoenix :+1: 

Back in business.

closed 2016-08-01 by jhunkeler