mperrin/webbpsf #140: In version 0.6 'normalize' is no longer an expected keyword argument in calcPSF

kjbrooks, 2016-10-11:
In testing out WebbPSF version 0.6, I found that when trying to make a simulation of the NRM mode for NIRISS, I can no longer normalize the PSF at the exit pupil (as described in the **PSF Normalization** section [here](https://pythonhosted.org/webbpsf/webbpsf.html)) because the key word argument 'normalize' is unexpected. 

Here is an example of what I get when I try to use 'normalize':

```
In [3]: niriss = webbpsf.NIRISS()

In [4]: niriss.pupil_mask = 'MASK_NRM'

In [5]: niriss.calcPSF(outfile='niriss_psf_exit.fits',normalize='exit_pupil')
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-6-e3bd53f6b57e> in <module>()
----> 1 niriss.calcPSF(outfile='niriss_psf_exit.fits',normalize='exit_pupil')

TypeError: calc_psf() got an unexpected keyword argument 'normalize'
```

mperrin, 2016-10-11:
This is related to #132. 

Looks like the underlying issue is `poppy.Instrument.calc_psf` isn't passing through `normalize` to `poppy.OpticalSystem.calc_psf`.

mperrin, 2016-10-11:
Fixed in this commit just pushed to poppy master: https://github.com/mperrin/poppy/commit/cf4bddeb184e4ef15ca4ec76c41abceb184d3db2

closed 2016-10-12 by josePhoenix