mperrin/webbpsf #150: NIRISS auto-pupil attempts to infer wavelength for CLEAR

josePhoenix, 2016-11-04:
An unanticipated consequence of #64 was that the `filter` setter now fails to guess the wavelength (when `auto_pupil` is enabled). (This was also reported by Kevin Volk.)

```
In [1]: %paste
import webbpsf
niriss = webbpsf.NIRISS()
niriss.filter = 'CLEAR'

## -- End pasted text --
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-1-98db0a3fdf08> in <module>()
      1 import webbpsf
      2 niriss = webbpsf.NIRISS()
----> 3 niriss.filter = 'CLEAR'

/Users/jlong/software/webbpsf/webbpsf/webbpsf_core.py in filter(self, value)
   1325
   1326         if self.auto_pupil:
-> 1327             wlnum = int(self.filter[1:4])
   1328             new_pupil_mask = self.pupil_mask # default no change
   1329             if wlnum >= 250:

ValueError: invalid literal for int() with base 10: 'LEA'
```

mperrin, 2016-11-04:
Yeah I saw that too in his message. Are you able to take care of this now or should I? I'm sitting here in the ICC watching mirrors move (sloooooowwwwly) so I could add the necessary special case for this now if you want.

josePhoenix, 2016-11-04:
Whoops, missed your message. I've got the fix in a new PR now, and emailed Kevin the workaround (set `niriss.auto_pupil = False`).

commit 780aedf1 (2016-11-04): Fix #150: unable to guess pupil when niriss.filter is CLEAR
closed 2016-11-04 by mperrin