mperrin/webbpsf #153: How does the NIRCam model handle filters in the pupil wheel?

josePhoenix, 2016-11-29:
When analyzing some CV3 test data, I was made aware of the fact that some filters are mounted in the pupil wheel and used with a wide filter in the filter wheel. In particular, F407N and F444W are used together.

Since WebbPSF doesn't pretend to model overall system throughput, and since the top of the F444W filter is pretty smooth in the region of F407N (as one might expect), I doubt it needs to be treated explicitly. However, a note in `jwst.rst` might not be amiss.

mperrin, 2016-11-29:
It handles it correctly so far as I know. Remember behind the scenes all the throughput files are derived from Pandora then normalized to leave out the total throughput. The Pandeia calculation of the system spectral response does properly include both the above filters, or at least that's my understanding. Could doubled check with Tim if you want to be really sure.

mperrin, 2016-11-29:
Pandeia not Pandora. For some reason iOS autocorrect doesn't know about the Greek goddess of brightness...

mperrin, 2017-08-11:
stale issue, no action needed.

closed 2017-08-11 by mperrin