mperrin/webbpsf #155: double check MIRI Lyot radius

mperrin, 2016-12-15:
In clearing out my overflowing inbox I came across the following from last August, which we never followed up on. _This is very low priority, I'm just making it an issue now so we remember to do it sometime - but not right now in the end of year rush._

Joseph: 
> Just realized we have two slightly different numbers for this in WebbPSF and the JWST Target Visibility Tool:
> https://github.com/spacetelescope/jwst_visibility/blob/master/jwst_visibility/gui.py#L980   (radius = 4.25 / 2 = 2.125 arcsec)
> https://github.com/spacetelescope/jwst_visibility/blob/master/jwst_visibility/gui.py#L980 (radius = 2.16 arcsec)
> I'm not sure we even know the radius to ~3 hundredths of an arcsec precision, but thought I'd check if either of you thought changes were needed. CCed Brendan because he was asking me about it.

Marshall: 
> The MIRI PASP paper on the coronagraph says 2.16” radius (http://arxiv.org/pdf/1508.02352v1.pdf at top of page 5) 
> But my comments in the webbpsf source code say "#diameter is 4.25 (measured) 4.32 (spec) supposedly 6 lambda/D”
which is consistent with the 2.125 number. However right now I can’t find the original source for that supposed measured value. 
> We may want to check with Dean Hines and see if he knows, and if not I’ll send a query to Anthony Boccaletti.

josePhoenix, 2016-12-15:
David Law might be the person to ask. I believe he was working on the MIRI SIAF inputs.

mperrin, 2016-12-15:
oh come on, I explicitly put that disclaimer at the top of the issue so you _would not spend any time on this now_ while dealing with your other deadlines!  ;-)

josePhoenix, 2016-12-15:
Haven't left the house yet, so it doesn't count!

mperrin, 2018-08-28:
Issues migrated to https://github.com/spacetelescope/webbpsf/issues. 
Please see the copies in that repo from now on.

closed 2018-08-28 by mperrin