mperrin/webbpsf #156: WebbPSF-WFIRST_Tutorial example misleading

rubab1, 2016-12-30:
The worked examples shows that 
`mono_psf = wfi.calc_psf(monochromatic=1.2e-6)`

has EE50 = 0.069" and FWHM = 0.138" whereas WebbPSF built in functions calculate approximate EE50 = 0.084" and FWHM = 0.105". 

`import webbpsf`
`import webbpsf.wfirst`
`wfi = webbpsf.wfirst.WFI()`
`mono_psf = wfi.calc_psf(monochromatic=1.2e-6)`
`webbpsf.measure_fwhm(mono_psf)`
`0.1047605318024962`
`getEE = webbpsf.measure_EE(mono_psf)`
`getEE([0.069,0.08,0.082,0.084,0.086])`
`array([ 0.43128342,  0.4948787 ,  0.50307424,  0.50997386,  0.51562519])`

The mismatch is significant in context. Apparently the mismatch is due to the display_profiles function using sparsely sampled radial profile and EE to estimate EE50, then setting FWHM = 2*EE50.

josePhoenix, 2017-05-26:
Hi @rubab1, @mperrin and I agree this isn't how the FHWM annotation should work. The implementation is not in WebbPSF, but rather in mperrin/poppy if you would like to open a Pull Request™.

P.S. Say hi to @meredith-durbin for me 😄

mperrin, 2018-08-28:
Issues migrated to https://github.com/spacetelescope/webbpsf/issues. 
Please see the copies in that repo from now on.

closed 2018-08-28 by mperrin