mperrin/webbpsf #157: NIRCam coronagraphy fails with mask=None

JarronL, 2017-03-02:
When performing coronagraphic simulations with NIRCam, if the mask is omitted (but the Lyot pupil is in place), then an image plane is no longer inserted at the mask location. This causes an indexing issue later down the road. It appears that [line 1010 of webbpsf_core.py ](https://github.com/mperrin/webbpsf/blob/master/webbpsf/webbpsf_core.py#L1010)
`            #optsys.add_image(poppy.ScalarTransmission(name='No Image Mask Selected!'), index=1)`
was commented out at some point. Uncommenting this line and changing to `index=2` fixes the issue for me.

mperrin, 2017-03-06:
Hmm, what's the use case for this?  I think my reasoning for commenting it out was that it would make the propagation faster to avoid two propagation transforms that end up being a null operation.  That would imply that a better fix would be to fix the indexing issue instead. Do you actually care about the existence of that empty image plane, or do you just want to be able to calculate Lyot pupil PSFs w/out a mask?   (For those I've been actually calculating off-axis PSFs myself, but I can see the benefit of taking the shortcut of mask=None)

JarronL, 2017-05-25:
Sorry,I thought that I had responded to this, but apparently not(?). 

I would like to calculate a PSF by setting mask=None and pupil='CIRCLYOT' (for instance) and don't have any restrictions on how this is accomplished in the propagation (either by my suggestion above or fixing the root of the index issue). I generally use this method for quickly creating a background PSF with a small footprint for quick saturation and bg sensitivity calculations. For these purposes, I found that calculating an off-axis PSF to be a little clunky and preferred the mask=None method (it's also computationally faster that dealing with the coronagraphic mask).

mperrin, 2018-04-27:
This was fixed at some point in the past. I don't recall in which commit. But I just double checked now and it is working as desired.

closed 2018-04-27 by mperrin