mperrin/webbpsf #161: Hardcoded pupil dimensions within JWST SI WFE models

josePhoenix, 2017-04-06:
When I tried swapping in OSIM pupils and OPD maps, it became clear there's code expecting a 1024x1024 array (lots of `IndexError: index 924 is out of bounds for axis 0 with size 162`).

Minor, for people using the provided OPD and pupil maps. But worth thinking about how the models could be refactored to compute things on the right sampling "just in time" (using Wavefront.coordinates, probably).

mperrin, 2017-04-06:
I'm not surprised. I put some thought into this when coding these but never fully tested this. At least some of the code should be using the input array sizes (see https://github.com/mperrin/webbpsf/blob/master/webbpsf/optics.py#L1064) but I'm not surprised that some other places slipped through the cracks.  

Can you paste an error log or at least which line that's coming from? The `MIRIFieldDependentAberrationAndObscuration` class definitely has hard coded 1024s, and the main `WebbFieldDependentAberration` class _only_ if the include_oversize option is set to True, which isn't the default.

josePhoenix, 2017-04-06:
Yeah, it's the MIRI one. The constructor for that WFE element has include_oversize set to True by default, as well. Good to know it's not everywhere :)

mperrin, 2017-04-06:
For the cross checks you're doing now, it would be a reasonable workaround to make a branch of the code that just invokes the regular `WebbFieldDependentAberration` instead of the special MIRI one.  Trying to draw in the field dependent obscuration is irrelevant if you're patching in OSIM pupil maps anyway.  So try just commenting out line 722 in `webbpsf_core.py` that invokes the MIRI aberration class.

josePhoenix, 2017-04-06:
Is the obscuration really not relevant for OSIM? Or are you saying if the obscuration were in the beam at that field point, Randal's pupil shapes would already include it?

mperrin, 2017-04-06:
The latter.

josePhoenix, 2017-04-06:
Side note, no source edits needed (god bless Python):

```
miri._si_wfe_class = webbpsf.optics.WebbFieldDependentAberration
```

mperrin, 2018-04-27:
This is related to the 2048 squared pupils discussed as part of #204 . @obi-wan76, this issue points out a couple places with hard-coded 1024 sizes for some of the other SIs, in particular MIRI.

mperrin, 2018-08-28:
Issues migrated to https://github.com/spacetelescope/webbpsf/issues. 
Please see the copies in that repo from now on.

closed 2018-08-28 by mperrin