mperrin/webbpsf #165: Include the updated filters files from Pandeia

mperrin, 2017-05-11:
Wow, I can't believe this, but it looks like a year ago when I re-generated all the throughput files from Pandeia, I never actually put the new ones in webbpsf_data_source on Central Store. So we are still using the old ones rather than the more updated throughput curves. Whoops. 

Not a huge deal - since Pandeia itself is just using monochromatic calculations and then doing the broadband normalizations itself. But this should get fixed before whenever we actually release 0.6 is. 

The new files should be generated with dev_utils/Normalized Throughputs from Pandeia.ipynb and put into place.

shanosborne, 2018-02-26:
This should be done - especially for the W2 filters!

mperrin, 2018-05-21:
Done, in webbpsf-data-master now and in the 0.7.0 rc zip files.

closed 2018-05-21 by mperrin