mperrin/webbpsf #166: poppy.test() fail on python 2.7

benjaminpope, 2017-06-27:
Hi Marshall,

I am hoping to use poppy for an aperture mask simulation I'm doing. I just ran poppy.test() on a brand new conda 4.3.21 python 2.7 install on Mac Sierra and it seems to fail on 3 tests unexpectedly. I have copied the full output below. Is this a known issue?

Using astropy 1.3.3, scipy 0.19.1, numpy 1.13.0, matplotlib 2.0.2. 

All the best,

Ben 

=================================== FAILURES ===================================
_____________________________ test_optic_resizing ______________________________

    @pytest.mark.skipif(
        (scipy is None),
        reason='No SciPy installed'
    )
    def test_optic_resizing():
        '''
        Tests the rescaling functionality of OpticalElement.getPhasor(),
        by first creating an optic with a small pixel scale and then
        creating an optic with a large pixel scale, and checking the returned
        phasor of each has the dimensions of the input wavefront.
        '''
    
        # diameter 1 meter, pixel scale 2 mm
        inputwf = poppy_core.Wavefront(diam=1.0, npix=500)
    
        # Test rescaling from finer scales: diameter 1 meter, pixel scale 1 mm
        test_optic_small=fits.HDUList([fits.PrimaryHDU(np.zeros([1000,1000]))])
        test_optic_small[0].header["PUPLSCAL"]=.001
        test_optic_small_element=poppy_core.FITSOpticalElement(transmission=test_optic_small)
>       assert(test_optic_small_element.getPhasor(inputwf).shape ==inputwf.shape )

/anaconda/lib/python2.7/site-packages/poppy/tests/test_core.py:348: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
/anaconda/lib/python2.7/site-packages/poppy/poppy_core.py:2315: in getPhasor
    return self.get_phasor(wave)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <poppy.poppy_core.FITSOpticalElement object at 0x111470e50>
wave = <poppy.poppy_core.Wavefront object at 0x111470cd0>

    def get_phasor(self,wave):
        """ Compute a complex phasor from an OPD, given a wavelength.
    
            The returned value should be the complex phasor array as appropriate for
            multiplying by the wavefront amplitude.
    
            Parameters
            ----------
            wave : float or obj
                either a scalar wavelength or a Wavefront object
    
            """
        #_log.info("Pixelscales for %s: wave %f, optic  %f" % (self.name, wave.pixelscale, self.pixelscale))
    
        if isinstance(wave, Wavefront):
            wavelength=wave.wavelength
        else:
            wavelength=wave
        scale = 2. * np.pi / wavelength.to(u.meter).value
    
        # set the self.phasor attribute:
        # first check whether we need to interpolate to do this.
        float_tolerance = 0.001  #how big of a relative scale mismatch before resampling?
        if self.pixelscale is not None and hasattr(wave,'pixelscale') and abs(wave.pixelscale -self.pixelscale)/self.pixelscale >= float_tolerance:
            _log.debug("Pixelscales: wave {}, optic {}" .format(wave.pixelscale, self.pixelscale))
            #raise ValueError("Non-matching pixel scale for wavefront and optic! Need to add interpolation / ing ")
            if hasattr(self,'_resampled_scale') and abs(self._resampled_scale-wave.pixelscale)/self._resampled_scale >= float_tolerance:
[... 142 lines cut ...]
E       TypeError: slice indices must be integers or None or have an __index__ method

/anaconda/lib/python2.7/site-packages/poppy/tests/test_fresnel.py:354: TypeError
___________________________ test_estimate_nprocesses ___________________________

    def test_estimate_nprocesses():
        """ Apply some basic functionality tests to the
        estimate nprocesses function.
        """
        osys = poppy_core.OpticalSystem("test")
        osys.addPupil(optics.CircularAperture(radius=1))
        osys.addPupil(optics.CircularAperture(radius=0.5))
        osys.addDetector(pixelscale=0.1, fov_arcsec=2.0)
    
        answer = utils.estimate_optimal_nprocesses(osys)
    
        #see if it's an int with a reasonable value
        assert type(answer) is int
    
>       assert answer > 0, "Estimated optimal nprocesses must be positive integer"
E       AssertionError: Estimated optimal nprocesses must be positive integer
E       assert 0 > 0

/anaconda/lib/python2.7/site-packages/poppy/tests/test_multiprocessing.py:105: AssertionError
========= 3 failed, 92 passed, 6 skipped, 1 xfailed in 170.51 seconds ==========

mperrin, 2017-06-27:
Hmm, thanks for bringing this to our attention. The first two errors are clearly due to the recent change in numpy handling of indices, but I thought we'd fixed all those cases already. The third test failure's a new one I don't recognize but I suspect it may be some related float/integer issue. 

Thanks for bringing this to our attention, and we will try to look into it more sometime soon.

josePhoenix, 2017-06-27:
I thought we'd taken care of these, too. @benjaminpope could you say which version of POPPY you're using?

Also, if it's the current version, would you mind opening an issue on mperrin/poppy instead? Thanks :)

josePhoenix, 2017-06-27:
Ah, they were fixed on `master` since the 0.5.1 release. I would recommend installing POPPY from the development sources (see http://pythonhosted.org/poppy/installation.html) until we get a chance to release 0.5.2 or 0.6.0.

benjaminpope, 2017-06-28:
Ah, thanks! Dev source worked. Was using standard pip distribution. Can close this issue then.

closed 2017-06-29 by benjaminpope