mperrin/webbpsf #167: Add the WFIRST "blue" (orange) filter

josePhoenix, 2017-06-27:
The WFIRST WFI has gained a shorter wavelength filter.

josePhoenix, 2017-06-27:
CC @meredith-durbin for reference info

meredith-durbin, 2017-06-27:
Passing the buck to @rubab1, since I got the filter curves from him

rubab1, 2017-06-28:
The most recent filter definitions (Cycle 6) preceded selection of the Orange filter. I agree that inclusion of the new filter soon is essential, when a new curve is officially available.

josePhoenix, 2017-07-07:
From Jeff Kruk:

> The transmission curve data are posted on the public WFIRST website:
https://wfirst.gsfc.nasa.gov/science/Instrument_Reference_Information.html
>
> If you scroll down to near the bottom of this page, the first link under 'Spreadsheets' has the full spreadsheet with all of the filters. This should be in the same format as the previous spreadsheets we have posted.
>
> We don't yet have updated zernike coefficients, but I will let you know when we have them.

rubab1, 2017-07-07:
@josePhoenix Thanks for the link, didn't know it was public already. Let's hope for a speedy delivery of the updated zernike coefficients :-)

mperrin, 2018-05-21:
Added by @robelgeda  as R062

closed 2018-05-21 by mperrin