mperrin/webbpsf #168: Data Files

benjaminpope, 2017-06-29:
Just used conda to install webbpsf - I get an environment error saying I need 0.6.0 data files to run it when I try to import (see error below).

But when I go to the link to get the data files, it looks like 0.5.0 are the only ones available?

 ***********  ERROR  ******  ERROR  ******  ERROR  ******  ERROR  ***********
 *                                                                          *
 *  WebbPSF requires several data files to operate.                         *
 *  These files could not be located automatically at this time, or this    *
 *  version of the software requires a newer set of reference files than    *
 *  you have installed.  For more details see:                              *
 *                                                                          *
 *           http://pythonhosted.org/webbpsf/installation.html              *
 *                                                                          *
 *  under "Installing the Required Data Files".                             *
 *  WebbPSF will not be able to function properly until the appropriate     *
 *  reference files have been downloaded to your machine and installed.     *
 *                                                                          *
 ****************************************************************************
---------------------------------------------------------------------------
EnvironmentError                          Traceback (most recent call last)
<ipython-input-1-777ca7dc744a> in <module>()
----> 1 import webbpsf

/Users/bjsp/code/webbpsf/webbpsf/__init__.py in <module>()
     89     # defined in either the environment or in webbpsf.cfg
     90     try:
---> 91         tmp, data_files_version = utils.get_webbpsf_data_path(data_version_min=DATA_VERSION_MIN, return_version=True)
     92         del tmp
     93     except (EnvironmentError, IOError):

/Users/bjsp/code/webbpsf/webbpsf/utils.pyc in get_webbpsf_data_path(data_version_min, return_version)
    228                 "for a link to the latest version.".format(
    229                     cur=version_contents,
--> 230                     min='{}.{}.{}'.format(*data_version_min)
    231                 )
    232             )

EnvironmentError: WebbPSF data package has version 0.5.0, but 0.6.0 is needed. See http://pythonhosted.org/webbpsf/installation.html#data-install for a link to the latest version.

mperrin, 2017-06-29:
Conda should be installing version 0.5 still. The version 0.6 is still not officially released (and the data files aren't public yet, pending various project sign-offs that have been much delayed for various reasons). Did you specifically tell conda to install the 0.6 version of webbpsf from github? If so I'd suggest not doing that just quite yet. :-)   You can install the 0.6 dev version of poppy just fine, though.

mperrin, 2017-08-11:
0.6 is now released, including the data files.

closed 2017-08-11 by mperrin