mperrin/webbpsf #170: Migration to ReadTheDocs

josePhoenix, 2017-08-11:
It appears that https://pythonhosted.org/ quietly removed the ability to upload new ZIPs of documentation. 😿 

ReadTheDocs.org is more featureful (in some ways), and more commonly used. The reason we weren't using it already is that (when we last checked) projects that depend on SciPy and NumPy wouldn't build correctly in their environment. This appears to be fixed now. However, we have to do a few things to work around the absence of the WebbPSF data files during documentation builds.

## Migration Checklist

  - [ ] Create project at readthedocs.org (@mperrin)
  - [ ] In Advanced Settings, check "Install your project inside a virtualenv using setup.py install"
  - [ ] Fill in `docs/conf.py` as "Python configuration file"
  - [ ] Fill in `docs/rtd-pip-requirements` as the "requirements file"
  - [ ] Check "Give the virtual environment access to the global site-packages dir." (This makes SciPy and friends available.)
  - [ ] Select CPython 3.x as the interpreter
  - [ ] Configure / verify the [GitHub webhook](http://docs.readthedocs.io/en/latest/webhooks.html#webhook-creation)

josePhoenix, 2017-08-11:
My test build (https://readthedocs.org/projects/webbpsf-josephoenix/builds/5820847/) failed due to the absence of `webbpsf-data`:

```
Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/webbpsf-josephoenix/envs/latest/local/lib/python3.5/site-packages/sphinx/cmdline.py", line 295, in main
    opts.warningiserror, opts.tags, opts.verbosity, opts.jobs)
  File "/home/docs/checkouts/readthedocs.org/user_builds/webbpsf-josephoenix/envs/latest/local/lib/python3.5/site-packages/sphinx/application.py", line 163, in __init__
    confoverrides or {}, self.tags)
  File "/home/docs/checkouts/readthedocs.org/user_builds/webbpsf-josephoenix/envs/latest/local/lib/python3.5/site-packages/sphinx/config.py", line 134, in __init__
    execfile_(filename, config)
  File "/home/docs/checkouts/readthedocs.org/user_builds/webbpsf-josephoenix/envs/latest/local/lib/python3.5/site-packages/sphinx/util/pycompat.py", line 129, in execfile_
    exec_(code, _globals)
  File "conf.py", line 88, in <module>
    __import__(setup_cfg['package_name'])
  File "/home/docs/checkouts/readthedocs.org/user_builds/webbpsf-josephoenix/envs/latest/local/lib/python3.5/site-packages/webbpsf-0.6.0rc1-py3.5.egg/webbpsf/__init__.py", line 91, in <module>
    tmp, data_files_version = utils.get_webbpsf_data_path(data_version_min=DATA_VERSION_MIN, return_version=True)
  File "/home/docs/checkouts/readthedocs.org/user_builds/webbpsf-josephoenix/envs/latest/local/lib/python3.5/site-packages/webbpsf-0.6.0rc1-py3.5.egg/webbpsf/utils.py", line 200, in get_webbpsf_data_path
    raise EnvironmentError("Environment variable $WEBBPSF_PATH is not set!")
OSError: Environment variable $WEBBPSF_PATH is not set!

Exception occurred:
  File "/home/docs/checkouts/readthedocs.org/user_builds/webbpsf-josephoenix/envs/latest/local/lib/python3.5/site-packages/webbpsf-0.6.0rc1-py3.5.egg/webbpsf/utils.py", line 200, in get_webbpsf_data_path
    raise EnvironmentError("Environment variable $WEBBPSF_PATH is not set!")
OSError: Environment variable $WEBBPSF_PATH is not set!
```

mperrin, 2017-08-11:
OK, we can work around that by disabling the check when the environment variable `$READTHEDOCS` is set. I'll do that this afternoon but first, lunch.

josePhoenix, 2017-08-11:
🍛 ❗️

mperrin, 2017-08-11:
- [ ] and file a support ticket to get the links on jwst.stsci.edu updated

closed 2017-08-11 by mperrin