mperrin/webbpsf #179: Update NIRCam LW SI WFE models to back out the F323N defocus

mperrin, 2017-12-12:
Coordinate w/ @JarronL to determine quantitatively what the right adjustments to make are.

mperrin, 2018-02-20:
@JarronL, remember this conversation back in December? You thought the OPD files I'm using (from CV3, in F323N) were pessimistic because some of the defocus is in that specific filter rather than the other NIRCam optics. Can you point me at any better information on this? What do you recommend in terms of the OPDs we use for LW?

JarronL, 2018-02-20:
While in Pasadena, you showed me a plot (I believe from Randal Telfer?) with the theoretical wavelength-dependent defocus (either from Zemax or Code V) compared to measured defocus during ISIM-level testing. My initial inclination is to say that the "predicted" focus values should be used in general, and then certain filters (such F323N) will have an additional power. I'm currently trying to dig up review documentation with filter-specific focus offsets.

mperrin, 2018-02-20:
[nircam_defocus_model_plot.pdf](https://github.com/mperrin/webbpsf/files/1742301/nircam_defocus_model_plot.pdf)

mperrin, 2018-02-20:
Hmm, let's try that as PNG instead: 
[image]

It's my plot, from data provided by Randal.  The blue curves are what is used in WebbPSF right now, as a function of wavelength (but not specifically as a function of different named elements in the pupil and filter wheels)>

JarronL, 2018-02-21:
Thanks. I'm currently looking at the cumulative data from CV2 on my machine, which shows the best-fit focus offsets for the various field points. It's a bit of a scatter. From what I can tell, the measured points in your plot correspond to the average of 13 different field points per module, each with a standard deviation of about 0.25mm. I dug up one technical report that concludes the measured focus offsets were consistent with the theoretical optical design with certain filters having additional offsets. I'll have to get back to you on which filters (definitely F323N) and by how much.

mperrin, 2018-04-25:
Hi @JarronL, wanted to follow up on this about the filter-dependent defocus. Shannon Osborne here is working on updating the instrument models so this would be very useful info to have. 

cc @shanosborne @obiwan76