mperrin/webbpsf #186: Redundant code in _get_aberrations

shanosborne, 2018-02-01:
Clean up redundant code (the error) in _get_aberrations function in core code

robelgeda, 2018-02-02:
Hi @shanosborne, what is the error in in ` _get_aberration`?

robelgeda, 2018-02-02:
Sounds good to me!

mperrin, 2018-02-20:
Closed by PR #189.   

(BTW, @shanosborne, a helpful feature on Github is that when you do a pull request that fixes an issue, you can include a string like "closes #186" as part of the Git commit message. Then Github will automatically know to mark the issue as closed when the PR is merged.  See [these GitHub docs](https://help.github.com/articles/closing-issues-using-keywords/) for details. Or you can come and manually comment that the issue should be closed, like this!)

closed 2018-02-20 by mperrin