mperrin/webbpsf #191: For coronagraph masks, add "bar_offset_y" for NIRCam. Add similar for the round occulters and MIRI as well.

mperrin, 2018-02-19:
As requested by @ariedel for use in Pandeia ETC coronagraph generation script

ariedel, 2018-02-19:
Use case is that Pandeia requires the PSF to be centered on a pixel; this currently requires us to snap to pixel positions rather than the actual requested XY location.

mperrin, 2018-02-19:
This turns out to be a real pain in the neck for MIRI, given the different rotated coordinate systems. Because of how the optics are implemented, the "natural" way to offset the FQPM is in V2V3, rotated by ~ 5 deg from the instrument detector X and Y (which is what Pandeia wants to work in). So even implementing a "simple" option to offset the focal plane mask by a given amount inevitable is going to involve some coordinate transforms.  Not hard, just more work. 

NIRCam's comparatively simple.

mperrin, 2018-04-27:
@ariedel, remind me where we left things with the coronagraph PSF generation script. I have this open issue about adding an option for shifting the masks around, and added code back in February to implement it. I'm not sure if any more is needed now? If you're happy with the coronagraph PSF generation script then good and I won't worry about any more effort on this issue.