mperrin/webbpsf #193: Travis docs builds have started failing for WebbPSF

mperrin, 2018-02-20:
See for instance [this job](https://travis-ci.org/mperrin/webbpsf/jobs/344042352):

```
loading intersphinx inventory from http://poppy-optics.readthedocs.io/objects.inv...
WARNING: failed to reach any of the inventories with the following issues:
WARNING: intersphinx inventory 'http://poppy-optics.readthedocs.io/objects.inv' not fetchable due to <class 'requests.exceptions.HTTPError'>: ('intersphinx inventory %r not fetchable due to %s: %s', 'http://poppy-optics.readthedocs.io/objects.inv', <class 'requests.exceptions.HTTPError'>, HTTPError(...))
[autosummary] generating autosummary for: api_reference.rst, available_opds.rst, fft_optimization.rst, gui.rst, help.rst, index.rst, installation.rst, intro.rst, jwst.rst, more_examples.rst, poppy.rst, references.rst, release.rst, relnotes.rst, sampling.rst, webbpsf.rst, wfirst.rst
[automodsumm] api_reference.rst: found 19 automodsumm entries to generate
Exception occurred:
  File "/home/travis/build/mperrin/webbpsf/astropy_helpers/astropy_helpers/extern/automodapi/automodsumm.py", line 496, in generate_automodsumm_docs
    doc = get_documenter(obj, parent)
TypeError: get_documenter() missing 1 required positional argument: 'parent'
The full traceback has been saved in /tmp/sphinx-err-0j8y9xrg.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!
Sphinx Documentation subprocess failed with return code 1
```

Same error is seen with the [poppy Travis docs builds as well](https://travis-ci.org/mperrin/poppy/jobs/343909728). 

@eteq any suggestions, please? Do we need to update to a new version of the astropy_helpers submodule?

JarronL, 2018-02-20:
Not sure if this is related, but Astropy v3.0 came out yesterday (or so). I had a problem today with pynrc readthedocs where the virtual environment grabbed the latest version of astropy (v3), which was breaking some of code compilations/initializations. I ended up enforcing a more strict version in the requirements.txt document to astropy==2 which solved my problem.

mperrin, 2018-02-20:
Thanks Jarron, I'll experiment with that and see!

JarronL, 2018-02-21:
Just to clarify, I modified my requirements.txt (which is specific to readthedocs). Looks like your equivalent is at https://github.com/mperrin/webbpsf/blob/master/docs/rtd-pip-requirements. I haven't dealt with Travis, though, so not sure how that all works.

mperrin, 2018-02-21:
Right, it's different for Travis. So far I've tried editing the .travis.yml file in the home directory and/or setup.py, but didn't figure out the right incantations yet.  I'm hoping that @robelgeda may be able to help out with digging into this since my plate is (as usual) overfull...

closed 2018-02-21 by mperrin

robelgeda, 2018-02-21:
@mperrin @JarronL I will look into this.

mperrin, 2018-02-21:
No need now @robelgeda  - see PR #194 by @sosey which fixed things a few minutes ago.

robelgeda, 2018-02-21:
Got it.

commit 66ed310c (2018-02-21): Merge pull request #194 from sosey/update_ah

updating astropy_helpers to v2.0.4. Fixes #193