mperrin/webbpsf #196: MIRI FND filter wavelength data doesn't go to short wavelengths

shanosborne, 2018-02-26:
Doesn't reach short enough wavelengths so the transmission goes to zero
[image]

mperrin, 2018-04-25:
Neither @shanosborne nor I can remember why we thought this was an issue.

mperrin, 2018-05-23:
The updated filter throughput file I made for 0.7.0 has a better wavelength range.

On the other hand it has some weird narrow spikes at longer wavelengths. Not sure if real but that's what's in Pandeia. In any case, we're now going to be synced up with the official reference data used in the ETC, so further investigation of this is not our job as part of WebbPSF dev. 

[image]

closed 2018-05-23 by mperrin