mperrin/webbpsf #198: GUI fails to launch: NSException

laurenmarietta, 2018-03-26:
Using the latest version of webbpsf (https://github.com/mperrin/webbpsf/commit/66ed310c33051a8e29a7ea9333aff75c2a00c38c) in Python 3.5, calling `webbpsf.gui()` results in this gnarly error:

```
2018-03-26 15:55:47.652 python[30499:1840209] -[NSApplication _setup:]: unrecognized selector sent to instance 0x7f89955c96d0
2018-03-26 15:55:47.655 python[30499:1840209] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplication _setup:]: unrecognized selector sent to instance 0x7f89955c96d0'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fffb89302cb __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fffcd74848d objc_exception_throw + 48
	2   CoreFoundation                      0x00007fffb89b1f04 -[NSObject(NSObject) doesNotRecognizeSelector:] + 132
	3   CoreFoundation                      0x00007fffb88a2755 ___forwarding___ + 1061
	4   CoreFoundation                      0x00007fffb88a22a8 _CF_forwarding_prep_0 + 120
	5   Tk                                  0x00000001197eec02 TkpInit + 471
	6   Tk                                  0x000000011976a2a9 Tk_Init + 1794
	7   _tkinter.cpython-35m-darwin.so      0x0000000119643114 Tcl_AppInit + 84
	8   _tkinter.cpython-35m-darwin.so      0x0000000119642deb _tkinter_create + 1115
	9   libpython3.5m.dylib                 0x000000010ba9d3e8 PyCFunction_Call + 280
	10  libpython3.5m.dylib                 0x000000010bb1845a PyEval_EvalFrameEx + 26138
	11  libpython3.5m.dylib                 0x000000010bb1c415 _PyEval_EvalCodeWithName + 2421
	12  libpython3.5m.dylib                 0x000000010bb11e1e PyEval_EvalCodeEx + 78
	13  libpython3.5m.dylib                 0x000000010ba7f71d function_call + 381
	14  libpython3.5m.dylib                 0x000000010ba57320 PyObject_Call + 96
	15  libpython3.5m.dylib                 0x000000010ba6c48d method_call + 141
	16  libpython3.5m.dylib                 0x000000010ba57320 PyObject_Call + 96
	17  libpython3.5m.dylib                 0x000000010bab3d0d slot_tp_init + 125
	18  libpython3.5m.dylib                 0x000000010bab0389 type_call + 297
	19  libpython3.5m.dylib                 0x000000010ba57320 PyObject_Call + 96
	20  libpython3.5m.dylib                 0x000000010bb180a1 PyEval_EvalFrameEx + 25185
	21  libpython3.5m.dylib                 0x000000010bb1c415 _PyEval_EvalCodeWithName + 2421
	22  libpython3.5m.dylib                 0x000000010bb1cd7f fast_function + 351
	23  libpython3.5m.dylib                 0x000000010bb17f23 PyEval_EvalFrameEx + 24803
	24  libpython3.5m.dylib                 0x000000010bb1c415 _PyEval_EvalCodeWithName + 2421
	25  libpython3.5m.dylib                 0x000000010bb11e1e PyEval_EvalCodeEx + 78
	26  libpython3.5m.dylib                 0x000000010ba7f71d function_call + 381
	27  libpython3.5m.dylib                 0x000000010ba57320 PyObject_Call + 96
	28  libpython3.5m.dylib                 0x000000010ba6c48d method_call + 141
	29  libpython3.5m.dylib                 0x000000010ba57320 PyObject_Call + 96
	30  libpython3.5m.dylib                 0x000000010bab3d0d slot_tp_init + 125
	31  libpython3.5m.dylib                 0x000000010bab0389 type_call + 297
	32  libpython3.5m.dylib                 0x000000010ba57320 PyObject_Call + 96
	33  libpython3.5m.dylib                 0x000000010bb180a1 PyEval_EvalFrameEx + 25185
	34  libpython3.5m.dylib                 0x000000010bb1c415 _PyEval_EvalCodeWithName + 2421
	35  libpython3.5m.dylib                 0x000000010bb1cd7f fast_function + 351
	36  libpython3.5m.dylib                 0x000000010bb17f23 PyEval_EvalFrameEx + 24803
	37  libpython3.5m.dylib                 0x000000010bb1c415 _PyEval_EvalCodeWithName + 2421
	38  libpython3.5m.dylib                 0x000000010bb1cd7f fast_function + 351
	39  libpython3.5m.dylib                 0x000000010bb17f23 PyEval_EvalFrameEx + 24803
	40  libpython3.5m.dylib                 0x000000010bb1c415 _PyEval_EvalCodeWithName + 2421
	41  libpython3.5m.dylib                 0x000000010bb11dc1 PyEval_EvalCode + 81
	42  libpython3.5m.dylib                 0x000000010bb43186 PyRun_InteractiveOneObject + 582
	43  libpython3.5m.dylib                 0x000000010bb42a8e PyRun_InteractiveLoopFlags + 222
	44  libpython3.5m.dylib                 0x000000010bb4297c PyRun_AnyFileExFlags + 60
	45  libpython3.5m.dylib                 0x000000010bb5b537 Py_Main + 3351
	46  python                              0x000000010ba41dc8 main + 216
	47  python                              0x000000010ba41ce4 start + 52
	48  ???                                 0x0000000000000001 0x0 + 1
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
``` 
I suspect this might have something to do with a backend... maybe?

JarronL, 2018-03-27:
This looks similar to issue #103.

I have found that in many cases ``matplotlib`` crashes when using the default backend (at least on Mac OS X and certain Linux distributions). Given the propensity for these crashes, it may be preferable to use a different graphics backend such as TkAgg. This can either be accomplished by setting ``matplotlib.use("TkAgg")`` after importing ``matplotlib`` or setting the default backend via your matplotlibrc file (https://matplotlib.org/users/customizing.html#the-matplotlibrc-file). I personally employ the latter option.

mperrin, 2018-03-27:
The GUI is problematic because of different backends on different platforms, yes. I concur with the recommendation to try different backends and see if that works more reliably for your setup. That said, the GUI is not being actively developed any more, and there are many options that are available in the Python API but not the GUI. So just be aware that it's fine to use the GUI but you will have more capabilities if you work directly with the programming interface.

laurenmarietta, 2018-03-27:
Great - declaring `matplotlib.use('TkAgg')` before importing webbpsf did the trick. 

It might be useful to add a clause to the GUI script that determines what matplotlib backend is actively being used, and asks the user to change to TkAgg if it detects a different backend (rather than running and crashing)? Though I hear that the GUI isn't being developed anymore, so maybe it's not worth the trouble.

Just a thought - Marshall, if active development of the GUI is something you think would be worthwhile to start up again, that's something I might be interested in taking up!

commit 5b9646f6 (2018-12-07): Various docs updates. Closes #114, closes #116, closes #181, closes #198, closes #216. (#248)

* Various docs updates. Closes #114, closes #116, closes #181, closes #198, closes #216.

Man, that's a very satisfying commit message to write.
closed 2018-12-07 by mperrin