mperrin/webbpsf #203: Change WFIRST Imager psf settings

ariedel, 2018-04-25:
The current setup for toggling the WFIRST WFI pupil mask is (at my request, admittedly) an attribute called 'pupil', which can be set to WFI._unmasked_pupil_path or WFI._masked_pupil_path to select an unmasked or cold pupil masked setup.

For all of the JWST instruments, this is actually controlled by an attribute called pupil_mask, which is set to a string (for instance: for NIRCam, "WEDGELYOT" sets the lyot mask)

Could I get the WFIRST WFI pupil mask selection changed to a JWST-style attribute-based system? Something where I could set:
`WFI.pupil_mask = "COLD_PUPIL"`
to get the cold pupil mask.

mperrin, 2018-04-25:
And it would auto-select based on filter name, the same way that the JWST NIRISS model handles auto-inserting the CLEARP obscuration.

mperrin, 2018-04-27:
@robelgeda do you think you could take a look at this request from @ariedel?

robelgeda, 2018-04-27:
@mperrin : On it.