mperrin/webbpsf #204: Updates to SI models based on CV3 and other SI testing data

mperrin, 2018-04-25:
Notes from discussion with Marcio and Shannon - cc @shanosborne, @obi-wan76

**SI defocus models**
 * NIRCam defocus vs wavelength. See #179 
 * Are there any measured & significant focus variations between the instruments, or are they all entirely confocal within the uncertainties? 
- [ ] @obi-wan76 will confer with Telfer about focus information

**Pupil shear and pupil distortions.** 
 * To the extent that pupil distortions are in the SIAF, this will be automatically handed by @shanosborne's new code for applying distortions.  
- [ ] Confirm with Telfer and Sahlmann that the SIAF distortion models includes all the info from CV2+CV3 for our purposes here. 

 * Same for the pupil rotations mostly. The default per each SI was taken from the SIAF a while ago, and hard-coded into webbpsf_core. 
- [ ] @shanosborne can double check the rotation values in webbpsf_core are still consistent with the latest numbers. 

 * Pupil shear/offsets. Doesn't matter for imaging modes with clear pupils. Does matter for modes with additional pupil obscurations inside the SIs, in particular the NIRCam and MIRI coronagraph modes.  See Table 19 of Aronstein's WFE report. 

- [ ] @shanosborne can update the SI `__init__` functions to set `options[pupil_shift_x]`, `options[pupil_shift_y]` to the values from the report. 

**SI WFE Zernikes** 
* Were there any changes to WFE maps over the past ~ year since @mperrin imported them into webbpsf? As of June 2017, Aronstein told Perrin that the WFE maps had not changed since the late 2016 delivery.  Probably no change since but should verify. 
- [ ] @mperrin and @obi-wan76 will re-verify/cross-check the WFE info and Zernike per SI. 
- [ ] Can assess/evaluate whether there would be value in using any of the as-built optical models to help further constrain the spatial variations. Probably not - there's a lot of unavoidable uncertainty in any pre flight WFE maps and it may not be worth additional effort for higher accuracy. 

**Pupil Map resolution for large FOV PSFs, and incoherent scattered light**
 * Marcio can put his existing info on very large FOV PSFs and scattered light into a report or JDox page.  Turns out JStans and Holler already have plans for such a JDox page, and we can assist as needed but otherwise let them do it. Lightsey is working to extend scattered light to MIRI wavelengths. 
 * (As discussed previously w/ Marcio and JStans, large FOV PSFs needs better sampling). Consider setting up a better supported mode for WebbPSF to use 2048 squared OPDs instead of 1024, for better sampling of the segment gaps (and thus better modeling of the diffraction spikes). 

- [ ] @mperrin, @obi-wan76 can collaborate to make it easier to use the 2048 squared OPDs. If the user selects a 2048 sized pupil, the OPDs should all be upsampled to match automatically. 

 * Incoherent scattered light
- [ ] Consider adding the incoherent scattered light as another sub-function within the 'detector effects' post-processing framework that @shanosborne is developing. It's not actually a detector effect but even so it seems like we could add this in a similar way. Add a broad but faint Gaussian halo component around the PSF. 

**Other**

- [ ] @obi-wan76 will review the rest of Aronstein's report for any other low-hanging fruit of value we can update.

obi-wan76, 2018-04-26:
From Randal regarding the defocus/wavelength function. 
"The best thing to do is to synthesize all of our ISIM CV and NIRCam CV tests into a consistent model.  I still haven’t written up my OTIS reports so I need to get that out the door first, but after that I will get back to this older stuff, that one is definitely on my list to do."

obi-wan76, 2018-04-26:
I'll look into my old CV3 analysis of the best focus positions. I used encircle energy fractions to estimate the best focus for CV3 observations and I was usually in good agreement with everyone else values, including Randal's PR results (I was doing a cross-checked analysis).  Maybe I can create my own defocus vs wavelength function.

mperrin, 2018-05-17:
@obi-wan76, @shanosborne could you help me update the checklist of tasks in this issue? I think a lot of these have been accomplished but it should be the two of you checking them off, not me. Thanks!

obi-wan76, 2018-05-21:
@mperrin how can edit this ticket to mark a task as complete? 
Thanks!

mperrin, 2018-05-21:
I believe you can just click on the checkboxes directly to mark them checked.  Let me know if that doesn’t work.

obi-wan76, 2018-05-21:
@mperrin I tried but I can't check the checkboxes. I see it as "unclickable". @shanosborne are you able to check the checkboxes? Perhaps is just me.

shanosborne, 2018-05-21:
No I also can't click the checkboxes

mperrin, 2018-05-24:
@obi-wan76 Did you ever decide if we should update the focus vs wavelength model for NIRCam at all? Right now since we’re basically ready to go with everything else, I’m inclined to leave any changes to the focus model for a future task. Yes?

obi-wan76, 2018-05-25:
@mperrin Yes. I think we need to wait for Randal's final report to include a field-dependent focus model. I got some DMS reprocess NIRCam CV3 data to work on my own corrections but we still have time to compile and upgrade to a final all-inclusive model (as a future task).